iqrfsdk/iqrf-daemon #5: JSON structure proposal

fmikulu, 2017-03-21:
Please comment JsonStructure.md: 
git@github.com:iqrfsdk/iqrf-daemon.wiki.git branch origin/JsonStructure 
Add examples to JsonStructureDpaExample.md in the same repo.
Lets work on it for a while in the branch and publis to master later to wiki:
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructure
Use syntax for colored JSON syntax e.g.:
```json
      "thermometer": {
        "addr": 0,
        "cmd": "READ",
        "hwpid": ffff,
        "temperature": 23.0
        "rawReq": 00.00.0a.00.ff.ff,
        "rawResp": 00.00.0a.80.00.00.00.00.17.70.01
      } 
```

fmikulu, 2017-03-21:
Well the colors doesn't seem nice 👎

Roman3349, 2017-03-21:
Your JSON is invalid. 

Valid JSON is below.

```JSON
      "thermometer": {
        "addr": 0,
        "cmd": "READ",
        "hwpid": "ffff",
        "temperature": 23.0,
        "rawReq": "00.00.0a.00.ff.ff",
        "rawResp": "00.00.0a.80.00.00.00.00.17.70.01"
      } 
```

fmikulu, 2017-03-21:
OK, thx 👍

spinarr, 2017-03-21:
JsonStructureDpaExamples.md added into the wiki repository - branch JsonStructure

fmikulu, 2017-03-22:
JsonStructure branch merged to master => published. Please invite other reviewers. Review could be done both by this discussion or a new dedicated issue. A branch of wiki pages is possible as well. Thx.

openhsorg, 2017-03-24:
tested

spinarr, 2017-04-03:
Feedback for DPA structure from the technical call:

- rename mid field since mid evokes module id instead of message id (message_id)

- distinguish between standard peripheral and standard device in ptype field

E.g. we are having Thermometer as a standard DPA peripheral and soon there will
be also standard DPA device Thermometer. Ptype field should distinguish that:

- std_per_thermometer
- std_dev_thermometer

fmikulu, 2017-04-14:
Please @spinarr review:
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructure-v1
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructureDpa-v1

spinarr, 2017-04-21:
- keep sequence request, confirmation, response
- keep nadr instead of node
- suggestion: type:raw-hdp (header parsed)

fmikulu, 2017-04-21:
Updated according previous comment:
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructure-v1
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructureDpa-v1

fmikulu, 2017-05-05:
The JSON structure basics proposed as described above, Other improvements and additions will be covered by dedicated issues.

closed 2017-05-05 by fmikulu