iqrfsdk/iqrf-daemon #7: MqttMessaging SSL cert file parameter

fmikulu, 2017-03-21:
Missing parameter for certification file. Seems as passable by:
MQTTAsync.h:
```h
	/** The file in PEM format containing the public digital certificates trusted by the client. */
	const char* trustStore;

	/** The file in PEM format containing the public certificate chain of the client. It may also include
	* the client's private key. 
	*/
	const char* keyStore;
```
It's FFS, shall be verified against appropriate broker

fmikulu, 2017-03-26:
https://docs.loraserver.io/loraserver/configuration/

ca certificate used by the api server (optional)
CA_CERT=

tls certificate used by the api server (optional)
TLS_CERT=

tls key used by the api server (optional)
TLS_KEY=

spinarr, 2017-03-26:
Adding support for the IQRF daemon to connect to secure MQTT broker.

We are missing some parameters in the configuration for MQTT channel:
https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/MqttMessaging.json

Taken from:

https://www.digitalocean.com/community/tutorials/how-to-install-and-secure-the-mosquitto-mqtt-messaging-broker-on-ubuntu-16-04

. . .
listener 1883 localhost

listener 8883
certfile /etc/letsencrypt/live/mqtt.example.com/cert.pem
cafile /etc/letsencrypt/live/mqtt.example.com/chain.pem
keyfile /etc/letsencrypt/live/mqtt.example.com/privkey.pem

listener 8883 sets up an encrypted listener on port 8883. This is the standard port for MQTT + SSL, often referred to as MQTTS. The next three lines, certfile, cafile, and keyfile, all point Mosquitto to the appropriate Let's Encrypt files to set up the encrypted connections.

mosquitto_pub -h mqtt.example.com -t test -m "hello again" -p 8883 --capath /etc/ssl/certs/ -u "sammy" -P "password"

fmikulu, 2017-03-26:
Implemented in 6ae39bd1f74549eca17e5e6dca87df28da5f3438 in branch origin/feature/issue/7/MqttCert
Please test it

spinarr, 2017-04-04:
1883 : MQTT, unencrypted
 8883 : MQTT, encrypted
 8884 : MQTT, encrypted, client certificate required
 8080 : MQTT over WebSockets, unencrypted
 8081 : MQTT over WebSockets, encrypted

The encrypted ports support TLS v1.2, v1.1 or v1.0 with x509 certificates and require client support to connect. In all cases you should use the certificate authority file mosquitto.org.crt to verify the server connection. Port 8884 requires clients to provide a certificate to authenticate their connection.

spinarr, 2017-04-04:
https://github.com/eclipse/mosquitto/blob/master/client/pub_client.c
https://github.com/eclipse/mosquitto/blob/master/client/client_shared.c

#ifdef WITH_TLS
	printf(" --cafile : path to a file containing trusted CA certificates to enable encrypted\n");
	printf("            communication.\n");
	printf(" --capath : path to a directory containing trusted CA certificates to enable encrypted\n");
	printf("            communication.\n");
	printf(" --cert : client certificate for authentication, if required by server.\n");
	printf(" --key : client private key for authentication, if required by server.\n");
	printf(" --ciphers : openssl compatible list of TLS ciphers to support.\n");
	printf(" --tls-version : TLS protocol version, can be one of tlsv1.2 tlsv1.1 or tlsv1.\n");
	printf("                 Defaults to tlsv1.2 if available.\n");
	printf(" --insecure : do not check that the server certificate hostname matches the remote\n");
	printf("              hostname. Using this option means that you cannot be sure that the\n");
	printf("              remote host is the server you wish to connect to and so is insecure.\n");
	printf("              Do not use this option in a production environment.\n")

spinarr, 2017-04-12:
Implemented in 4ada2bd in branch origin/feature/issue/7/MqttCert
Confirmed

fmikulu, 2017-04-12:
Merged at f578686abebcbec2c961fdbf4a0764f07ac51c92

closed 2017-04-12 by fmikulu