iqrfsdk/iqrf-daemon #8: Create mode service for the daemon and enable UDP channel in service mode

fmikulu, 2017-03-24:
Imported from: https://iqrfsdk.atlassian.net/browse/SDKST-30?filter=-1
Can you describe required  feature?

spinarr, 2017-03-25:
Since UDP channel goes directly to SPI or CDC interface other channels must be disabled at that time.
It is advised to introduce different modes in which daemon operates. E.g.

- operational (UDP channel is dissabled, MQ and MQTT enabled)

- service (UDP channel is enabled, MQ and MQTT dissabled - we have IDE IQMesh Network Manager in mind as a servise tool via UDP - bonding/unbonding/uploading/backup)

- forwarding (all trafic from MQ or/and MQTT is forwarded to UDP port - IDE can be used for packet analysis)

Switching among modes may be done via iqrf json config commands - yet to be specified.

message category (ctype) : config
message type (type) : configReq

{
  "config": {
  	"mid": 1,
  	"configReq": {
           "mode":"service",
           ...
        } 
    }
}

fmikulu, 2017-03-26:
The switching among modes could be done as proposed. On the other hand wouldn't be better to switch it directly from IDE? There are periodic messages sent from IDE "GET_GW_STATUS" so just a simple flag added to the message would do the job easily.

fmikulu, 2017-03-26:
Proposed configuration:
```json
{
  "RemotePort": 55000,
  "LocalPort": 55300,
  "Mode": "service",
  "ServiceTimeoutMillis": 30000
}
```
**Mode** operational || service || forwarding
**ServiceTimeoutMillis** timeout to switch back to operational from service

Now the UdpMessaging implementation switches automatically to the **service** immediately when a traffic is detected from the IDE. It stays in the **service** up to  the timeout elapsing. This feature may be removed as soon as the configuration switching will be done according one of the proposals.

spinarr, 2017-03-26:
message category (ctype) : config
message type (type) : configReq

can be also issued via mq channel - e.g. via iqrfapp locally

spinarr, 2017-03-27:
One more clarification:

IQRF IDE is not sending "GET_GW_STATUS" all the time.
There are states in IQRF IDE when this packet is not sent.

fmikulu, 2017-03-31:
Lets review proposal.

Modalities to switch:
- Json msg via MqttMessagin:  `{config: {confiqReq ... }}`
- Json msg via MqMessaging `{config: {confiqReq ... }}`
- Dedicated cmdl like via MqMessaging, just to keep it simple, e.g. **> iqrfapp mode service**
- Special command from IDE - optionaly supported by IDE in future

Modes:
- **operational** UDP, MQ, MQTT, ... **works in parallel**, UDP is ordinary source of DPA requests
- **service** UDP channel has exclusive access, MQ, MQTT, ... dissabled - we have IDE IQMesh Network Manager in mind as a service tool via UDP - bonding/unbonding/uploading/backup
- **forwarding** all trafic from MQ, MQTT, ... is forwarded to UDP - IDE can be used for packet analysis

The implementation will allow UDP to be easily replaceable in its service/forwarding role by other messaging components. In this way the IDE can use different methods of messaging or any  other service tool could be used in future.

spinarr, 2017-03-31:
ok

fmikulu, 2017-04-01:
The modes **Operational, Service, Forwarding** are implemented as proposed. Switching is still TBD. However first tests shows that forwarded messages presentation in the IDE Terminal Log Window is questionable.  See in table bellow lines 4, 7:

Line 1 - 3 is DpaRequest to node 1 sent directly from IDE
Line 4 - 6 is DpaRequest to node 1 forwarded to IDE
Line 7 - 8 is DpaRequest to coordinator (node 0) forwarded to IDE
Line 4 - 6 is DpaRequest to coordinator (node 0) sent directly from IDE

It seems the IDE would be aware about forwarding mode and present forwarded Requests in a different way.

Line  | Time | Rx/Tx  | Length | DataHex | DPA | Error
--- | --- | --- | --- | --- | --- | ---
1 | 10:14:13.755 | Tx | 6 | 01.00.0A.00.FF.FF. | Request | 
2 | 10:14:13.849 | Rx | 11 | 01.00.0A.00.FF.FF.FF.68.00.03.00. | Confirmation | 
3 | 10:14:13.922 | Rx | 11 | 01.00.0A.80.00.00.00.69.15.51.01. | Response | 
4 | 10:17:40.092 | **Rx** | 6 | 01.00.0A.00.FF.FF. | **Asynchronous** | 
5 | 10:17:40.147 | Rx | 11 | 01.00.0A.00.FF.FF.FF.68.00.03.00. | Confirmation | 
6 | 10:17:40.220 | Rx | 11 | 01.00.0A.80.00.00.00.68.16.68.01. | Response | 
7 | 10:20:00.073 | **Rx** | 6 | 00.00.0A.00.FF.FF. | **Notification** | 
8 | 10:20:00.128 | Rx | 11 | 00.00.0A.80.00.00.00.68.1A.A0.01. | Response | 
9 | 10:20:03.959 | Tx | 6 | 00.00.0A.00.FF.FF. | Request | 
10 | 10:20:04.016 | Rx | 11 | 00.00.0A.80.00.00.00.68.1A.A0.01. | Response |

fmikulu, 2017-04-01:
Implemented Mode switch by iqrfapp:
```
>> conf Forwarding
>> conf Service
>> conf Operational
```

spinarr, 2017-04-23:
Suggestion before closing the issue:

- do not leave udp channel enabled by default in operational mode
(it is not fully functional since we are not able to set timeout for dpa messages)

fmikulu, 2017-05-07:
Udp channel disabled in operational mode 361463815d8f5f13a1744a9147a47423e8e2b86d

closed 2017-05-07 by fmikulu