iqrfsdk/iqrf-daemon #22: Finalize Configuration parameters

fmikulu, 2017-04-04:
Enhance  IqrfInterface.json:
- tr modul(type, mid, os version build)
- dpa version (22x, 30x)
- communication mode(std, lp)

Enhance UdpMessaging.json:
- ID
- FW version
- MAC (automatic?)
- Hostname
- IP
- Public IP
- Connection?
- IQRF OS (BUILD) - taken from IqrfInterface.json?

spinarr, 2017-04-04:
- IqrfInterface.json

Example taken from IDE
TR module type: DCTR-72Dx, MID: 81102D58, OS version: 4.00D (08B1), MCU type: PIC 16LF1938, 

There is also example using CDC library for getting this info, but we can start from having 
parameters defined in configuration IqrfInterface.json.

https://github.com/iqrfsdk/jlibcdc/blob/develop/jlibcdc-examples/src/main/java/com/microrisc/cdc/examples/Info.java

- UdpInterface.json

ID: propose to use globally unique IEEE EUI-64 identifier
mac1(byte1)-mac2-mac3-ff-fe-mac4-mac5-mac6(byte6)

FW: version of the iqrf-daemon x.y.z notation

MAC: we can start via configuration file, could be set via config web page we are preparing

Hostname: /etc/hostname, could be set via config web page we are preparing

IP: we can start via configuration file, could be set via config web page we are preparing

Connection: ETH, WiFi, LTE, also via config web page?

IQRF params take from IqrfInterface.json if possible.

May some parameters (Gateway EUI, MAC, Hostname, FW) be better located in main config.json file?

Roman3349, 2017-04-05:
**Bash commands for getting:**

- MAC address of default route:
```bash
cat /sys/class/net/$(ip route show default | awk '/default/ {print $5}')/address
```
- Hostname:
```bash
hostname -f
```
- IPv4 and IPv6 addresses:
```bash
hostname -I
```

fmikulu, 2017-04-07:
It seems little more complicated.

**Config Web Page** - I consider it as a  static information - just an editor for  *json files. There isn't now a runtime connection (in future we can provide web GUI, with a new channel, e.g. WebsocketChannel and send receive data from/to daemon). So I don't consider **Config Web Page** as a consumer now. If we need a runtime parameter update or read, it seems more reasonable to use existing MqChannel as already used for switching Operational/Service/Forwarding. 

Some parameters are not related to a consumer, so do we need them at all? If there is a purpose I cannot see it, so please provide a clue

Parameter | Origin  | Origin details | Consumers | Question
--- | --- | --- | --- | --- 
tr modul(type, mid, os version build) | DpaRequest | OS Read | UDP CMD 0x01, 0x11 | 
dpa version (22x, 30x) | DpaRequest | Peripheral Enum from Coordinator | ? | What to do with it? Is origin correct?
communication mode(std, lp) | DpaRequest | Peripheral Enum from Coordinator | ? | What to do with it? Is origin correct?
ID | Generated | From MAC | UDP CMD 0x01 | 
FW version | Hardcoded | Can be generated from CMakeLists.txt | UDP CMD 0x01 | 
MAC | SystemCall |  | UDP CMD 0x01 | 
IP | SystemCall |  | UDP CMD 0x01 | 
Public IP | Configuration | UdpMessaging.json | UDP CMD 0x01 | 
Hostname | Configuration | UdpMessaging.json | ? | What to do with it?
Connection | Configuration | UdpMessaging.json | ? | What to do with it?
TCP/IP Stack Version | Configuration | UdpMessaging.json | UDP CMD 0x01 | 
Net BIOS Name | Configuration | UdpMessaging.json | UDP CMD 0x01 |

spinarr, 2017-04-12:
clibdpa is consumer of these:
- dpa version (22x, 30x)
- communication mode (std, lp)

udp cmd 0x01
- maybe hostname can be sent in net bios name parameter
- connection will not be used

fmikulu, 2017-04-12:
Then the **clibdpa** would start in an indifferent mode (if possible) and expect some initialization step with dpa and mode specification. It could be done by sending (if possible in the indifferent mode) `Peripheral Enum` DPA request to the Coordinator or just from configuration (which may be inconsistent with real HW). 

The parameter **hostname** is not so important now, so let have it it configuration just now. later we can use reverse DNS or something else.

fmikulu, 2017-04-26:
Use IqrfSpiChannel::setCommunicationMode() to switch the speed according 5x or 7x

fmikulu, 2017-04-26:
Use IqrfSpiChannel::setCommunicationMode to switch the speed according 7x resp 5x

spinarr, 2017-07-28:
*UDP*
- there is need to sorted initial IDE page after IDE gets connected to the daemon
- there are params(table above) which needs to be introduced (initialized) from daemon configuration
- daemon configuration should be therefore extended to cover these params

*IQRF*
- dpa version 22x, 30x
- std x lp mode
- default: std and 30x

fmikulu, 2017-08-25:
@spinarr 
The issue is quite stale. Can you revise with respect to clibdpa https://github.com/iqrfsdk/clibdpa/issues/13 and https://github.com/iqrfsdk/clibdpa/issues/11 what shall be done?

spinarr, 2017-08-25:
- wrt clibdpa there should be a selection of communication mode: STD, LP
- wrt clibdpa supported dpa version for 1.0.0 will be 30x (no need for user selection for now)
- wrt udp nothing has changed

fmikulu, 2017-08-26:
Split to two simpler #60 and #61

closed 2017-08-26 by fmikulu