iqrfsdk/iqrf-daemon #24: Create install for iqrfapp

fmikulu, 2017-04-05:
Add install to CMakeLists.txt
It is related to #15 - seems reasonable to install together with configuration iqrfapp.json

fmikulu, 2017-04-06:
Implemented at 5b633bd13e7dd7de2630e02eb411aac39ad005d4
Please verify if works as expected

spinarr, 2017-04-24:
pi@raspberrypi:~/IQRF/iqrf-daemon $ sudo iqrfapp Raw 01 00 06 03 ff ff
Raw 01 00 06 83 1e 00 00 32  STATUS_NO_ERROR

- ok

pi@raspberrypi:~/IQRF/iqrf-daemon $ sudo iqrfapp conf Operational
Usage
  iqrfapp [<perif> <num> <command>]

  iqrfapp raw <buffer>

  iqrfapp conf [Operational|Forwarding|Service]

Example
  iqrfapp
  iqrfapp Thermometer 1 READ
  iqrfapp LedG 0 PULSE
  iqrfapp LedR 1 PULSE
  iqrfapp Raw 01 00 06 03 ff ff
  iqrfapp Raw 01.00.06.03.ff.ff
  iqrfapp conf Operational

- is that switched?

pi@raspberrypi:~/IQRF/iqrf-daemon $ sudo iqrfapp conf Service
Usage
  iqrfapp [<perif> <num> <command>]

  iqrfapp raw <buffer>

  iqrfapp conf [Operational|Forwarding|Service]

Example
  iqrfapp
  iqrfapp Thermometer 1 READ
  iqrfapp LedG 0 PULSE
  iqrfapp LedR 1 PULSE
  iqrfapp Raw 01 00 06 03 ff ff
  iqrfapp Raw 01.00.06.03.ff.ff
  iqrfapp conf Operational

- switched?

fmikulu, 2017-04-25:
Sorry, I didn't test it properly outside interactive mode. Fixed at b7481c736cf61d188f5175c976df2289460522d4

@spinarr can you try it again?

fmikulu, 2017-05-05:
Working

closed 2017-05-05 by fmikulu