iqrfsdk/iqrf-daemon #28: Adding daemon version and build date

spinarr, 2017-04-12:
Adding daemon version and build date via cmake and build/packaging script.
Version and build date can be written into log file at the daemon start.

fmikulu, 2017-04-18:
Implemented at dda8333aae6e2eb104dbbc9d541cadeac22d8bd2
The parameters DAEMON_VERSION + BUILD_TIMESTAMP are traced during startup in the logging output:
```
====================================================
DAEMON_VERSION="v1.0.0" BUILD_TIMESTAMP="Mon 04/17/2017 21:24:36.70"
====================================================
```
The parameters can be set during cmake phase. The example is at: https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/build64.bat

```bat
set ver=v1.0.0
set tms="%date% %time%"

rem //launch cmake to generate build environment
pushd %builddir%
cmake -G "Visual Studio 14 2015 Win64" -Dclibdpa_DIR:PATH=%clibdpa% -Dcutils_DIR:PATH=%cutils% -Dclibcdc_DIR:PATH=%clibcdc% -Dclibspi_DIR:PATH=%clibspi% -DDAEMON_VERSION:STRING=%ver% -DBUILD_TIMESTAMP:STRING=%tms% %currentdir%
popd
```

This solution allows passing ver + tms as parameters to the build scripts and automate it via CI. However it is necessary to adapt all build scripts.

Another solution is to set everything from CMakeLists.txt (easy change)

Please consider what do you preferred @spinarr @Roman3349

spinarr, 2017-04-21:
Build scripts will be adjusted.

fmikulu, 2017-04-21:
@Roman3349 what do you think? Updating the scripts would allow automation of packaging: creating the git tag, setting version according the tag, setting date of build, etc. On the other hand we have to maintain a lot of scripts now. Updating CMakeLists.txt would be just one place.

fmikulu, 2017-04-23:
CMake tests DAEMON_VERSION + BUILD_TIMESTAMP.

If empty => sets its own values hardcoded in [CMakeLists.txt](https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/CMakeLists.txt).

It still keeps possibility to set up the values by build* scripts as shown at: [build64.bat](https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/build64.bat)

closed 2017-04-23 by fmikulu