iqrfsdk/iqrf-daemon #29: MqttMessaging.json unused timeout parameter

fmikulu, 2017-04-17:
There is unused parameter "Timeout" in the MqttMessaging configuration. Review hard-coded values and make them configurable if applicable.

fmikulu, 2017-04-21:
There are these possible time values to be parametric defaulted to:
```cpp
int keepAliveInterval = 20; //special msg sent to keep connection alive
int connectTimeout = 5; //waits for accept from broker side
int minReconnect = 1; //waits to reconnect when connection broken
int maxReconnect = 64; //waits time *= 2 with every unsuccessful attempt up to this value 
```
@spinarr does it make sense to parametrise any of them? If yes I will turn **"Timeout"** to some of them or add another or just omit **"Timeou**

spinarr, 2017-04-21:
We expose all and close issue. We leave default values in configuration file.

fmikulu, 2017-04-21:
Done at dc66795f9d5722e628d6f95cc348b1edbd182395

closed 2017-04-21 by fmikulu