iqrfsdk/iqrf-daemon #34: Create parsers for DDC-SE/RE

fmikulu, 2017-04-21:
"type" : "custom_per_ddcse"
"type" : "custom_per_ddcre"

DDC-SE-01: HWPID=0x001E
DDC-RE-01: HWPID=0x002E
DDC-SE-01 & DDC-RE-01: HWPID=0x003E

----------- DDC-SE-01
Request: PNUM = 0x20, PCMD = 0x31
Response:
 PData[0]=Temperature at [C]
 PData[1]=Photoresistor value 0x00-0xFF
 PData[2]=Potentiometer value 0x00-0xFF
 returns ERROR_FAIL when error reading temperature

FRC=0xC0, returns temperature at C, 127 for 0 C, 0x80 for error reading temperature
FRC=0xC1, returns photoresistor value, returns 1 instead of 0
FRC=0xC2, returns potentiometer value, returns 1 instead of 0

----------- DDC-RE-01
Request: PNUM = 0x20, PCMD = 0x32
 PData[0]=Relay1Ctrl
 PData[1]=Relay2Ctrl

Response:
 PData[0]=Relay1State
 PData[1]=Relay2State

RelayCtrl: 0=Switch the Relay off, 1=Switch the Relay on, Other=Do not control the Relay
RelayState: 0=Relay was off, 1=Relay was on

spinarr, 2017-05-08:
This custom handlers will be replaced by standard sensor and actuator.

ctype:dpa
type:std-dev-sen
...

&

ctype:dpa
type:std-dev-act
...

Standardization work in progress.

fmikulu, 2017-05-12:
Covered by #49

closed 2017-05-12 by fmikulu