iqrfsdk/iqrf-daemon #36: Solve excessive wdog logging

fmikulu, 2017-04-22:
There's too much regular trace records from "wdog pet"

fmikulu, 2017-04-22:
The logging is produced by the Scheduler as the watchdog petting is fired as a task. The cleanest solution is to introduce Logger named channels and enable/disable according the channels and verbosity. The implementation will be done for v2.0.0. The only possibilities for now are to comment out annoying logging for all tasks or to do a hack - conditional logging according the task content. BTW: the tasks are logged  out during processing anyway. @spinarr: what's your opinion?

spinarr, 2017-04-22:
ok, we can leave it for v2.0.0, it is fine

fmikulu, 2017-04-22:
Disabled logs from Scheduler, lets see how it's going a8fcd4a69095409e0e91e3227c24268c2ea2ab90

closed 2017-04-22 by fmikulu