iqrfsdk/iqrf-daemon #41: Adapt iqrf_startup_pm to JsonStructuteDpa-v1

fmikulu, 2017-05-01:
It is not just about PrfPulseMeter, but it is necessary to cope with situations when DpaTask and derived classes are created from ctor. Some mandatory type, nadr, etc parameters are missing when encoded to JsonDpaResponse.

fmikulu, 2017-06-23:
Please review it @michal-konopa 

Implemented by 85b92becd9d5b880e10318393d529bb21304bfed
See changes in: https://github.com/iqrfsdk/iqrf-daemon/blob/master/applications/iqrf_startup_pm/iqrf_startup_pm/configuration/ClientServicePm.json mainly new item `DpaRequestJsonPattern` - it controls what is returned in response (shall be according JsonStructuteDpa-v1):

```json
{
  "Implements": "IService",
  "Instances": [
    {
      "Name": "ClientServicePm",
      "Messaging": "MqttMessaging",
      "Serializers": [
        "JsonSerializer"
      ],
      "Properties": {
        "FrcPeriod": 5,
        "SleepPeriod": 30,
        "Pulsemeters": [ 1, 2 ],
        "DpaRequestJsonPattern": {
          "ctype": "dpa",
          "type": "Pulsemeter",
          "cmd": "PULSE_READ",
          "hwpid": "ffff",
          "timeout": 10000,
          "request": "",
          "request_ts": "",
          "confirmation": "",
          "confirmation_ts": "",
          "response": "",
          "response_ts": ""
        }
      }
    }
  ]
}```

fmikulu, 2017-07-28:
See in my previous comment. Could be similar for ProtocolBridge

spinarr, 2017-07-28:
Looks fine, but I do not have HW to test it.
Project is still alive so we might test it later on in September.

closed 2017-07-28 by spinarr