iqrfsdk/iqrf-daemon #52: iqrf daemon hang - after ERROR_TIMEOUT

msalajka, 2017-06-14:
[gamma_log.txt](https://github.com/iqrfsdk/iqrf-daemon/files/1075214/gamma_log.txt)

spinarr, 2017-06-14:
Is it via USB CDC IQRF or SPI interface?

fmikulu, 2017-06-14:
It seems that something is quite slow. If you compare response_ts & confirmation_ts it takes tenths of milliseconds, but response_ts is 2 - 3 sec delayed. Explicit timeout is 10 sec. I can see PNUM 0x0C UART and PCMD 0x02 read/write - is it reason of such delay?
It is obvious that two requests and their responses are interleaved. It could be some bug in queuing. We need to prepare some stress tests to verify daemon resilience in such situations.

msalajka, 2017-06-15:
IqrfInterface.json

{
  "IqrfInterface": "/dev/ttyACM0",
  "DpaHandlerTimeout": 10000
}

msalajka, 2017-06-15:
how to replicate: i just powered off the sensors for some period of time... but when i turned on the sensors again, daemon is still "frozen..."

spinarr, 2017-06-17:
Timing is correct:

req: 01.00.0c.02.ff.ff.**fe**.01.42.00.03.75.31.75.33.75.32.01.86
reqt: 2017-06-14.12:34:00.2078
cont: 2017-06-14.12:34:00.28517
rest: 2017-06-14.12:34:02.646513

**fe** in DPA request is ReadTimeout ... waiting for response at the end device.

There could be test to adjust ReadTimeout to shorten the waiting interval
and see if the bug is related to this.

Will do more tests.

spinarr, 2017-06-18:
#!/bin/bash
# Script for testing MQTT channel on Linux machine
# mosquitto_pub is using MQTT local broker to communicate to IQRF daemon
# Tested on Raspberry PI 3, Raspbian Lite
# Tested on AAEON UP, UbiLinux

cnt=1
adr=1

echo "sending request to pulse red led"

while [ true ]
do
	echo "req n.$cnt adr n.$adr"

	mosquitto_pub -t "Iqrf/DpaRequest" -m "{\"ctype\":\"dpa\",\"type\":\"raw\",\"msgid\":\"$cnt\",\"timeout\":3000,\"request\":\"$adr.00.0c.02.ff.ff.fe.01.42.00.03.75.31.75.33.75.32.01.86\",\"request_ts\":\"\",\"confirmation\":\".\",\"confirmation_ts\":\"\",\"response\":\".\",\"response_ts\":\"\"}"

	cnt=$((cnt+1))

	adr=$((adr+1))
	if [ $adr -gt 2 ]
	then
		adr=1
	fi
	printf -v adr "%x" "$adr"

        echo "req n.$cnt adr n.$adr"

	mosquitto_pub -t "Iqrf/DpaRequest" -m "{\"ctype\":\"dpa\",\"type\":\"raw\",\"msgid\":\"$cnt\",\"timeout\":3000,\"request\":\"$adr.00.0c.02.ff.ff.fe.01.42.00.03.75.31.75.33.75.32.01.86\",\"request_ts\":\"\",\"confirmation\":\".\",\"confirmation_ts\":\"\",\"response\":\".\",\"response_ts\":\"\"}"

        cnt=$((cnt+1))

        adr=$((adr+1))
        if [ $adr -gt 2 ]
        then
                adr=1
        fi
        printf -v adr "%x" "$adr"

	sleep 10
done

Keep running ...
Q to clarify: why do I get 2 responses at once on mqtt?
I sent 2 requests at once and that is ok testwise.

Checking timing in 2 responses and seems to be ok.
Why I do not get first response after 3s and another after 6s on mqtt?

spinarr, 2017-06-18:
I do test only with coordinator running at this stage.

spinarr, 2017-06-19:
- it working fine if we are sending requests outside of daemon
(you find script in scripts folder in the daemon repository if you want to try for yourself)
- it gets frozen when sending the requests from the scheduler
- tests have been done only with coordinator running

spinarr@Linux:~$ mosquitto_sub -t Iqrf/DpaResponse
{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "7531",
    "timeout": 10000,
    "request": "01.00.0c.02.ff.ff.fe.01.42.00.03.75.31.75.33.75.32.01.86",
    "request_ts": "2017-06-18 20:50:00.1065",
    "confirmation": "01.00.0c.02.ff.ff.ff.2e.01.09.01",
    "confirmation_ts": "2017-06-18 20:50:00.50894",
    "response": "",
    "response_ts": "",
    "status": "ERROR_PROMISE_TIMEOUT"
}
{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "7534",
    "timeout": 10000,
    "request": "02.00.0c.02.ff.ff.fe.01.42.00.03.75.34.75.33.75.32.cd.86",
    "request_ts": "2017-06-18 20:50:10.391968",
    "confirmation": "",
    "confirmation_ts": "",
    "response": "",
    "response_ts": "",
    "status": "ERROR_TIMEOUT"
}
{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "7531",
    "timeout": 10000,
    "request": "01.00.0c.02.ff.ff.fe.01.42.00.03.75.31.75.33.75.32.01.86",
    "request_ts": "2017-06-18 20:52:00.1223",
    "confirmation": "01.00.0c.02.ff.ff.ff.2e.01.09.01",
    "confirmation_ts": "2017-06-18 20:52:00.51532",
    "response": "",
    "response_ts": "",
    "status": "ERROR_PROMISE_TIMEOUT"
}
{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "7534",
    "timeout": 10000,
    "request": "02.00.0c.02.ff.ff.fe.01.42.00.03.75.34.75.33.75.32.cd.86",
    "request_ts": "",
    "confirmation": "",
    "confirmation_ts": "",
    "response": "",
    "response_ts": "",
    "status": "ERROR_TIMEOUT"
}

That is all we receive then it is frozen.

fmikulu, 2017-06-19:
It seems like combination of two independent bugs. Fixing is in progress ...

fmikulu, 2017-06-22:
Daemon tried to exit by unsatisfied Watchdog in case of long DPA transactions. But there was a problem with pending transaction in DpaHandler and another bug causing a mutex deadlock in such situation.

The handler has a new method for aborting pending transaction. It is necessary to get the newest version of clibdpa: https://github.com/iqrfsdk/clibdpa/commit/aa5cdc7f70e8a0309b6d402df6db24e9e1301425

Deadlock during shutdown solved at 4e9c2cb5602a4cfbbe5756886f0392d26b3f9934

For longer DPA transactions it is necessary to disable watchdog by set time to zero or prolong it to appropriately in configuration file `config.json`:

```json
  "WatchDogTimeoutMilis": 10000,
```

@msalajka and @spinarr please review and test and let me now if it works as expected.

spinarr, 2017-07-20:
Looks good now.

closed 2017-07-20 by spinarr