iqrfsdk/iqrf-daemon #56: ProtocolBridge integration

fmikulu, 2017-08-04:
Unexpected exit() during normal ProtocolBridge operation.

We can observe in log:

```
03-08-2017 17:34:05.80105 {DBG} "ProtocolBridgeClientService.cpp" ln:400 confirmInvisibleMeters()
Response:
01 00 26 83 14 00 00 50 05 03 02 02 00 01       ..&....P...... 

03-08-2017 17:34:21.513686 {DBG} "DaemonController.cpp" ln:668 stop()
```
The timestamps shows obvious discontinuity of cca 16 seconds and thus hang up of the method: `ProtocolBridgeClientService::confirmInvisibleMeters()`

According attached logs it can be caused by abort of waiting transaction:
[ProtocolBridgeClientService.cpp#L416](https://github.com/iqrfsdk/iqrf-daemon/blob/feature/ProtocolBridgeClientService/applications/protocol_bridge/ProtocolBridgeClientService/ProtocolBridgeClientService.cpp#L416)

I can see inappropriate timers setting:
[configuration/config.json#L4](https://github.com/iqrfsdk/iqrf-daemon/blob/feature/ProtocolBridgeClientService/applications/protocol_bridge/startup/configuration/config.json#L4)
and:
[configuration/ProtocolBridgeClientService.json#L19](https://github.com/iqrfsdk/iqrf-daemon/blob/feature/ProtocolBridgeClientService/applications/protocol_bridge/startup/configuration/ProtocolBridgeClientService.json#L19)
If PB sleeps then the DpaTransaction can be aborted by WD

Try to prolong WD timer at least to 20 sec

Another issue is that this kind of abort blocks exit process.
 
[exit.txt](https://github.com/iqrfsdk/iqrf-daemon/files/1199855/exit.txt)
[ok.txt](https://github.com/iqrfsdk/iqrf-daemon/files/1199856/ok.txt)

fmikulu, 2017-08-04:
Blocked exit covered by #57

spinarr, 2017-08-04:
- according to traces in the log it never leaves confirmInvisibleMeters()
- watchdog timer set to 30s

closed 2017-09-15 by spinarr