iqrfsdk/iqrf-daemon #58: request: scheduler - run once after reboot

msalajka, 2017-08-11:
Hello,

It may be useful to be able to run dpa command with scheduler once after the restart as cron can do.

~~~
EXTENSIONS
       These special time specification "nicknames" which replace the 5 initial time and date fields, and are prefixed with the '@' character, are supported:

       @reboot    :    Run once after reboot.
       @yearly    :    Run once a year, ie.  "0 0 1 1 *".
       @annually  :    Run once a year, ie.  "0 0 1 1 *".
       @monthly   :    Run once a month, ie. "0 0 1 * *".
       @weekly    :    Run once a week, ie.  "0 0 * * 0".
       @daily     :    Run once a day, ie.   "0 0 * * *".
       @hourly    :    Run once an hour, ie. "0 * * * *".
~~~

the command i want to run after restart is, for example, discovery.

{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "discovery",
    "timeout": 10000,
    "request": "00.00.00.07.ff.ff.07.09",
    "request_ts": "",
    "confirmation": ".",
    "confirmation_ts": "",
    "response": ".",
    "response_ts": ""
}

m.

fmikulu, 2017-08-11:
It is a good idea to launch some DPA after reboot. I like the nicknames. I will implement it ASAP.
Thanks for contribution.

fmikulu, 2017-08-18:
Implemented as proposed:
```cpp
  //These special time specification "nicknames" which replace the 5 initial time and date fields,
  //and are prefixed with the '@' character, are supported :
  //  @reboot : Run once after reboot.
  //  @yearly : Run once a year, ie.  "0 0 0 0 1 1 *".
  //  @annually : Run once a year, ie.  "0 0 0 0 1 1 *".
  //  @monthly : Run once a month, ie. "0 0 0 0 1 * *".
  //  @weekly : Run once a week, ie.  "0 0 0 * * * 0".
  //  @daily : Run once a day, ie.   "0 0 0 * * * *".
  //  @hourly : Run once an hour, ie. "0 0 * * * * *".
  //  @minutely : Run once a minute, ie. "0 * * * * * *".
```
To schedule in this way, use in Schedule.json e.g:
```json
  "TasksJson": [
    {
      "time": "@reboot",
      "service": "BaseServiceForMQTT1",
      "message": {
        "ctype":"dpa",
        "type":"raw",
        "msgid":"1",
        "timeout":1000,
        "request":"00.00.06.03.ff.ff",
        "request_ts":"",
        "confirmation":".",
        "confirmation_ts":"",
        "response":".",
        "response_ts":""
      }
    }
]
```

fmikulu, 2017-08-18:
It is in da113be1165896215f81e4acb0083b0725ec5a1e master 
Will be part of next release V0.8.0

closed 2017-08-18 by fmikulu