iqrfsdk/iqrf-daemon #59: STATUS_NO_ERROR even if Dpa interface is not working

msalajka, 2017-08-11:
i guess, it's bug. daemon returns STATUS_NO_ERROR even if dpa interface
is not working...

i removed iqrf usb gw from the usb port while the daemon was running.

~~~
1-08-2017 11:36:00.569019 {DBG} "MqttMessaging.cpp" ln:391 delivered()
Message delivery confirmeddt="34"
11-08-2017 11:36:00.569800 {DBG} "MqttMessaging.cpp" ln:425 onSend()
Message sent successfuly
11-08-2017 11:36:00.574802 {DBG} "BaseService.cpp" ln:87 handleMsgFromMessaging()
==================================
Received from MESSAGING:
7b 0a 20 20 20 20 22 63 74 79 70 65 22 3a 20 22 {....."ctype":."
64 70 61 22 2c 0a 20 20 20 20 22 74 79 70 65 22 dpa",....."type"
3a 20 22 72 61 77 22 2c 0a 20 20 20 20 22 6d 73 :."raw",....."ms
67 69 64 22 3a 20 22 63 6f 32 2d 32 22 2c 0a 20 gid":."co2-2",..
20 20 20 22 74 69 6d 65 6f 75 74 22 3a 20 31 30 ..."timeout":.10
30 30 30 2c 0a 20 20 20 20 22 72 65 71 75 65 73 000,....."reques
74 22 3a 20 22 30 32 2e 30 30 2e 30 63 2e 30 32 t":."02.00.0c.02
2e 66 66 2e 66 66 2e 66 65 2e 30 31 2e 34 32 2e .ff.ff.fe.01.42.
30 30 2e 30 33 2e 37 35 2e 33 31 2e 37 35 2e 33 00.03.75.31.75.3
33 2e 37 35 2e 33 32 2e 30 31 2e 38 36 22 2c 0a 3.75.32.01.86",.
20 20 20 20 22 72 65 71 75 65 73 74 5f 74 73 22 ...."request_ts"
3a 20 22 22 2c 0a 20 20 20 20 22 63 6f 6e 66 69 :."",....."confi
72 6d 61 74 69 6f 6e 22 3a 20 22 2e 22 2c 0a 20 rmation":.".",..
20 20 20 22 63 6f 6e 66 69 72 6d 61 74 69 6f 6e ..."confirmation
5f 74 73 22 3a 20 22 22 2c 0a 20 20 20 20 22 72 _ts":."",....."r
65 73 70 6f 6e 73 65 22 3a 20 22 2e 22 2c 0a 20 esponse":.".",..
20 20 20 22 72 65 73 70 6f 6e 73 65 5f 74 73 22 ..."response_ts"
3a 20 22 22 0a 7d                               :."".}

11-08-2017 11:36:00.577658 {ERR} "DaemonController.cpp" ln:72 executeDpaTransactionFunc()
Dpa interface is not working
11-08-2017 11:36:00.580437 {DBG} "MqttMessaging.cpp" ln:258 sendTo()
Sending to MQTT: topic="iqrf/dparesponse"
7b 0a 20 20 20 20 22 63 74 79 70 65 22 3a 20 22 {....."ctype":."
64 70 61 22 2c 0a 20 20 20 20 22 74 79 70 65 22 dpa",....."type"
3a 20 22 72 61 77 22 2c 0a 20 20 20 20 22 6d 73 :."raw",....."ms
67 69 64 22 3a 20 22 63 6f 32 2d 32 22 2c 0a 20 gid":."co2-2",..
20 20 20 22 74 69 6d 65 6f 75 74 22 3a 20 31 30 ..."timeout":.10
30 30 30 2c 0a 20 20 20 20 22 72 65 71 75 65 73 000,....."reques
74 22 3a 20 22 30 32 2e 30 30 2e 30 63 2e 30 32 t":."02.00.0c.02
2e 66 66 2e 66 66 2e 66 65 2e 30 31 2e 34 32 2e .ff.ff.fe.01.42.
30 30 2e 30 33 2e 37 35 2e 33 31 2e 37 35 2e 33 00.03.75.31.75.3
33 2e 37 35 2e 33 32 2e 30 31 2e 38 36 22 2c 0a 3.75.32.01.86",.
20 20 20 20 22 72 65 71 75 65 73 74 5f 74 73 22 ...."request_ts"
3a 20 22 22 2c 0a 20 20 20 20 22 63 6f 6e 66 69 :."",....."confi
72 6d 61 74 69 6f 6e 22 3a 20 22 22 2c 0a 20 20 rmation":."",...
20 20 22 63 6f 6e 66 69 72 6d 61 74 69 6f 6e 5f .."confirmation_
74 73 22 3a 20 22 22 2c 0a 20 20 20 20 22 72 65 ts":."",....."re
73 70 6f 6e 73 65 22 3a 20 22 22 2c 0a 20 20 20 sponse":."",....
20 22 72 65 73 70 6f 6e 73 65 5f 74 73 22 3a 20 ."response_ts":.
22 22 2c 0a 20 20 20 20 22 73 74 61 74 75 73 22 "",....."status"
3a 20 22 53 54 41 54 55 53 5f 4e 4f 5f 45 52 52 :."STATUS_NO_ERR
4f 52 22 0a 7d                                  OR".}

11-08-2017 11:36:00.586791 {DBG} "MqttMessaging.cpp" ln:391 delivered()
Message delivery confirmeddt="35"
11-08-2017 11:36:00.588038 {DBG} "MqttMessaging.cpp" ln:425 onSend()
Message sent successfuly
~~~

m.

fmikulu, 2017-08-14:
It is a bug - shall be reported exactly back to the code handling the transaction result.

fmikulu, 2017-08-17:
Fixed by https://github.com/iqrfsdk/clibdpa/commit/a83824b9c908cc1d051efb1048dea91408d5f273
and b80fc32a7e25680053725a443a014226eb5bb605
It is in master and will be part of V0.8.0

closed 2017-08-18 by fmikulu