iqrfsdk/iqrf-daemon #60: Configuration via UDP

fmikulu, 2017-08-26:
It is derived from #22
Enhance UdpMessaging.json:

- ID
- FW version
- MAC (automatic?)
- Hostname
- IP
- Public IP
- Connection?
- IQRF OS (BUILD) - taken from IqrfInterface.json?

ID: propose to use globally unique IEEE EUI-64 identifier
mac1(byte1)-mac2-mac3-ff-fe-mac4-mac5-mac6(byte6)

FW: version of the iqrf-daemon x.y.z notation

MAC: we can start via configuration file, could be set via config web page we are preparing

Hostname: /etc/hostname, could be set via config web page we are preparing

IP: we can start via configuration file, could be set via config web page we are preparing

Connection: ETH, WiFi, LTE, also via config web page?

IQRF params take from IqrfInterface.json if possible.

May some parameters (Gateway EUI, MAC, Hostname, FW) be better located in main config.json file?

    there is need to sorted initial IDE page after IDE gets connected to the daemon
    there are params(table above) which needs to be introduced (initialized) from daemon configuration
    daemon configuration should be therefore extended to cover these params

fmikulu, 2017-08-27:
Partly implemented in daemon: 64f65de9e33336432fa4efe8976114b57fde8654 and cutils: https://github.com/iqrfsdk/cutils/commit/21c995b6dae09bca2e5e2c1bfc04dc234fd0dda0

It is implemented for WIN only, LIN is ASAP

Regarding ID - @spinarr  proposal is:
_ID: propose to use globally unique IEEE EUI-64 identifier
mac1(byte1)-mac2-mac3-ff-fe-mac4-mac5-mac6(byte6)_
But in documentation example is: _GW type e.g.: GW - ETH - 02A_ So I used _iqrf_daemon_

All other values are set reasonably, but there are still some unsolved values:
- TCP / IP Stack version e.g. : 5.42
- Net BIOS Name e.g. : iqrf_xxxx (15 characters)
- Public IP address e.g. : 213.214.215.120

@spinarr Is it important to provide it? If yes, then I will add it as Cfg pars to: https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/UdpMessaging.json

```cpp
void UdpMessaging::getGwIdent(ustring& message)
{
  //1. - GW type e.g.: GW - ETH - 02A
  //2. - FW version e.g. : 2.50
  //3. - MAC address e.g. : 00 11 22 33 44 55
  //4. - TCP / IP Stack version e.g. : 5.42
  //5. - IP address of GW e.g. : 192.168.2.100
  //6. - Net BIOS Name e.g. : iqrf_xxxx (15 characters)
  //7. - IQRF module OS version e.g. : 3.06D
  //8. - Public IP address e.g. : 213.214.215.120

  const char* crlf = "\x0D\x0A";

  std::basic_ostringstream<char> ostring;
  ostring << crlf <<
    "iqrf-daemon" << crlf <<
    m_daemon->getDaemonVersion() << crlf <<
    m_udpChannel->getListeningMacAddress() << crlf <<
    "0.00" << crlf << //TODO
    m_udpChannel->getListeningIpAddress() << crlf <<
    "iqrf_xxxx" << crlf << //TODO
    m_daemon->getOsVersion() << "(" << m_daemon->getOsBuild() << ")" << crlf <<
    m_udpChannel->getListeningIpAddress() << crlf; //TODO

  std::string resp = ostring.str();
  ustring res((unsigned char*)resp.data(), resp.size());
  message = res;
}
```

fmikulu, 2017-08-27:
UdpChannel gets MAC address for both WIN and LIN
https://github.com/iqrfsdk/cutils/commit/497d4d0d8fa1e4ad132de3c9289ab06f21d6bcc7

spinarr, 2017-08-29:
1) 
GW type e.g.: GW - ETH - 02A So I used iqrf_daemon ... **ok**

2) 
TCP / IP Stack version e.g. : 5.42 ... **no**
Net BIOS Name e.g. : iqrf_xxxx (15 characters) ... **hostname can be added to udp config**
Public IP address e.g. : 213.214.215.120 ... **ip address can be added to udp config**

3)
Should we also have unique ID regarding iqrf_daemon instances?
This may not be related to UDP but in general design if the IQRF GW is based on iqrf_daemon.
How is GW identified? Globally unique IEEE EUI-64 identifier still looks as good choice.

ID: propose to use globally unique IEEE EUI-64 identifier
mac1(byte1)-mac2-mac3-ff-fe-mac4-mac5-mac6(byte6)
or
mac1(byte1)-mac2-mac3-coordinatorMID1-coordinatorMID2-mac4-mac5-mac6(byte6)

MID is 4B so we would have to transform them into 2B, just idea then we could have
unique ID per instance since instance is related to one IQRF coordinator.

spinarr, 2017-09-04:
Clarify:

ID in IQRF IDE is created this way for User gateways (iqrf-daemon): 0x20 MAC4 MAC5 MAC6
and this is fine, so no need to extend UDP protocol.

IMHO, iqrf-daemon should have unique ID per instance, but this is not related to UDP and
IQRF IDE.

fmikulu, 2017-09-15:
Unique ID solved by #71

closed 2017-09-15 by fmikulu