iqrfsdk/iqrf-daemon #61: Enhance IqrfInterface congfiguration

fmikulu, 2017-08-26:
It is derived from #22
Enhance IqrfInterface.json:

- tr modul(type, mid, os version build)
- dpa version (22x, 30x)
- communication mode(std, lp)

Example taken from IDE
TR module type: DCTR-72Dx, MID: 81102D58, OS version: 4.00D (08B1), MCU type: PIC 16LF1938,

There is also example using CDC library for getting this info, but we can start from having
parameters defined in configuration IqrfInterface.json.

https://github.com/iqrfsdk/jlibcdc/blob/develop/jlibcdc-examples/src/main/java/com/microrisc/cdc/examples/Info.java

Use IqrfSpiChannel::setCommunicationMode to switch the speed according 7x resp 5x

IQRF

- dpa version 22x, 30x
- std x lp mode
- default: std and 30x

wrt clibdpa there should be a selection of communication mode: STD, LP
wrt clibdpa supported dpa version for 1.0.0 will be 30x (no need for user selection for now)

fmikulu, 2017-08-27:
DpaHandler set to STD/LP according IqrfInterface.json confuguration parameter CommunicationMode STD/LP implemented by 2003c8998714f5a5d39fc27ba24b101a091729e8

DPA version 22x 30x user selection is not required now so it is skipped

tr modul(type, mid, os version build) is solved in #62

fmikulu, 2017-08-27:
@spinarr  please review  2003c8998714f5a5d39fc27ba24b101a091729e8  if it implements expected behaviour

spinarr, 2017-08-29:
looks good

closed 2017-09-15 by fmikulu