iqrfsdk/iqrf-daemon #63: Add support for asynchronous DPA msgs

spinarr, 2017-08-29:
- space for initial discusion
- initial idea is to select communication channels (MQ/MQTT) to which async msgs should be transmitted in the configuration

fmikulu, 2017-09-09:
Passing AsyncDpaResponse shall be configurable (to where). It can be filtered according parameters in the response.

Possibly pending ordinary DpaTransaction shall be interruptible (async handling) and continue if possible.

This issue is related to #67

fmikulu, 2017-09-10:
# Proposal
AsyncDpaResponse is handled by dedicated instance of BaseService.

Example of BaseService.json:
```json
{
  "Implements": "IService",
  "Instances": [
    {
      "Name": "BaseServiceForMQ",
      "Messaging": "MqMessaging",
      "Serializers": [
        "SimpleSerializer",
        "JsonSerializer"
      ],
      "Properties": {
      }
    },
    {
      "Name": "BaseServiceForMQTT1",
      "Messaging": "MqttMessaging1",
      "Serializers": [
        "JsonSerializer"
      ],
      "Properties": {
        "AsyncDpaResponse": true
      }
    }
  ]
}
```
It means the instance BaseServiceForMQTT1 owns exclusive AsyncDpaResponse handling. An attempt  to register other instance fails with an exception and the first instance keeps the exclusivity.

spinarr, 2017-09-11:
AsyncDpaResponse could be named AsyncDpaMessage since both Req and Resp can come as async message.

So it is not possible to send async message to all instances that are marked and async true?

spinarr, 2017-09-11:
Currently clibdpa handles async message only if there is no dpa transfer in progress.
If there is dpa transfer then message is discarded, no buffering for later processing. 
Could be improved.

fmikulu, 2017-09-11:
Buffering AsyncDpaMessage => agree - an issue to clibdpa

AsyncDpaMessage non exlusive to all flagged async=true instances - it is possible - if required can be done in this way.

fmikulu, 2017-09-11:
Parsing AsyncDpaMessage in a similar way like in DpaExtension seems quite complex. In the scope of V1 all AsyncDpaMessages are sent towards Messaging instances as DpaRaw only.

fmikulu, 2017-09-15:
This is just handling of AsyncDpaMessage.

The requirement:
`Pending ordinary DpaTransaction shall be interruptible (async handling) and continue if possible.`
Is part of #67

spinarr, 2017-09-16:
- vote for **AsyncDpaMessage** in BaseService.json  and inlude it in all instances, default as false
- async message is not propagated via MQ channel (see TODO)
- handling of async messages will be improved in clibdpa https://github.com/iqrfsdk/clibdpa/issues/17 

Looks good.

fmikulu, 2017-09-16:
AsyncDpaMessage is propagated to all channesl even MQ.

The only problem is that SimpleSerializer doesn't implement it as we consider to drop SimpleSerializer at all.

So AsyncDpaMessage handling is simplified to use just the first parser to encode the message.

If the BaseService has more parsers, only the first one  (lets say default) is used e.g:
```json
    {
      "Name": "BaseServiceForMQ",
      "Messaging": "MqMessaging",
      "Serializers": [
        "SimpleSerializer",
        "JsonSerializer"
      ],
      "Properties": {
      }
    }
```
So switch the order in the BaseService.json configuration and AsyncDpaMessage encoded by JsonSerializer shall be sent to MQ:

```json
      "Serializers": [
        "JsonSerializer",
        "SimpleSerializer"
      ],
```
Note it wasn't tested yet :-)

spinarr, 2017-09-16:
Yes, it is working, very good.
Shall we rename configuration flag AsyncDpaResponse?

fmikulu, 2017-09-16:
OK, I will rename it.

spinarr, 2017-09-18:
Renamed https://github.com/iqrfsdk/iqrf-daemon/commit/203ec6bfc08dd85344cfec47d13112b61e18d804

closed 2017-09-18 by spinarr