iqrfsdk/iqrf-daemon #64: Create template for custom extensions - services

spinarr, 2017-08-29:
Basic template showing how to extend iqrf-daemon with custom logic.

fmikulu, 2017-10-05:
The example added via d78607face6976e0c1054d3e8c62c29f27b82c36

The scenario is (similar to pulsemetrs):

- Service example sends periodically FRC to catch connected thermometers.
- When  a thermometer is detected, it is read to get temperature and commanded to sleep for some time.
- When all configured thermometer detected FRC periodic command is stopped

The example covers:
Usage of PrfThermometer, PrfFrc, PrfOs, Sheduler, MqttMessaging, JsonSerializer, ...

It was derived from iqrf_setup_pm. It is maybe still too much complex. Definitely it needs to be commented reasonably. 

An alternative example could be (or maybe another one) a super simple with PrfRaw, just interpreting LED on/of from iqrfapp.

@spinarr please have a look and give me a feedback if it fulfills your requirement

spinarr, 2017-10-09:
I would start with super simple regarding dpa:

- use dpa raw periodically reading thermometer node1 (period scheduler)

- period defined from config file during boot and with possibility to overwrite the period over messaging 
  by custom json request message

- read temperature send over messaging with custom json response message

We will show:

- custom dpa logic
- custom json messages 

And we need reasonable comments for others to understand main concepts.

spinarr, 2017-10-17:
First example done, check README in examples!

closed 2017-10-17 by spinarr