iqrfsdk/iqrf-daemon #65: IQRFAPP output

spinarr, 2017-09-04:
**RIGHT**

sudo iqrfapp "{\"ctype\":\"dpa\",\"type\":\"raw\",\"msgid\":\"1\",\"timeout\":1000,\"request\":\"01.00.06.03.ff.ff\",\"request_ts\":\"\"}"

Sending DPA request to pulse red led on the node 1
{
    "ctype": "dpa",
    "type": "raw",
    "msgid": "1",
    "timeout": 1000,
    "request": "01.00.06.03.ff.ff",
    "request_ts": "2017-09-04 12:30:24.637778",
    "response": "01.00.06.83.02.00.00.40",
    "status": "STATUS_NO_ERROR"
}

**WRONG** when **confirmation** is included in json request

sudo iqrfapp "{\"ctype\":\"dpa\",\"type\":\"raw\",\"msgid\":\"1\",\"timeout\":1000,\"request\":\"01.00.06.03.ff.ff\",\"request_ts\":\"\",\"**confirmation**\":\".\"}"

ubilinux@ubilinux:~/IQRF/github/iqrf-daemon/scripts$ ./mq-iqrfapp-blink.sh 
Sending DPA request to pulse red led on the node 1
**(NO OUTPUT)**

spinarr, 2017-09-14:
sudo iqrfapp "{"ctype":"dpa","type":"raw","msgid":"1","timeout":1000,"request":"01.00.06.03.ff.ff","request_ts":"","confirmation":"."}"

in iqrf-daemon.log

14-09-2017 20:02:02.623348 {WAR} "MqChannel.cpp" ln:249 sendTo()
writeMq() failed: GetLastError="90"

fmikulu, 2017-09-17:
cutils https://github.com/iqrfsdk/cutils/commit/ac28d0b43d387c77addc3be9058c43639737054a
iqrf-daemon a8cf52ed003296ec589023c9ef359a010dcd3fb9
Please retest

closed 2017-09-20 by spinarr