iqrfsdk/iqrf-daemon #67: Getting module info

spinarr, 2017-09-05:
Is it possible to postpone getting information about DCTR module info?

There is reset of DCTR module after restart/start iqrf-daemon and DCTR module
sends async message after DCTR is reseted. 

DPA request is sent right after start and it interferes with async packet from DCTR
after reset, see log below:

Transfer status: created
05-09-2017 16:28:44.372372 {DBG} "DpaTransfer.cpp" ln:341 ProcessStatus()
{LEAVE} 
05-09-2017 16:28:44.372531 {DBG} "DpaHandler.cpp" ln:56 SendDpaMessage()
<<<<<<<<<<<<<<<<<<
Sent to DPA interface: 
00 00 02 00 ff ff                               ......          

05-09-2017 16:28:44.372771 {DBG} "IqrfSpiChannel.cpp" ln:174 sendTo()
Sending to IQRF SPI: 
00 00 02 00 ff ff  
05-09-2017 16:28:44.374007 {DBG} "IqrfSpiChannel.cpp" ln:197 sendTo()
status.isDataReady="0x0" status.dataNotReadyStatus="0x0" 
05-09-2017 16:28:44.374132 {DBG} "IqrfSpiChannel.cpp" ln:201 sendTo()
Sleep for a while ... 
05-09-2017 16:28:44.474317 {DBG} "IqrfSpiChannel.cpp" ln:177 sendTo()
Trying to sent: 1.2
05-09-2017 16:28:44.475375 {DBG} "IqrfSpiChannel.cpp" ln:197 sendTo()
status.isDataReady="0x0" status.dataNotReadyStatus="0x0" 
05-09-2017 16:28:44.475422 {DBG} "IqrfSpiChannel.cpp" ln:201 sendTo()
Sleep for a while ... 
05-09-2017 16:28:44.575629 {DBG} "IqrfSpiChannel.cpp" ln:177 sendTo()
Trying to sent: 1.3
05-09-2017 16:28:44.576871 {DBG} "IqrfSpiChannel.cpp" ln:197 sendTo()
status.isDataReady="0x0" status.dataNotReadyStatus="0x0" 
05-09-2017 16:28:44.576967 {DBG} "IqrfSpiChannel.cpp" ln:201 sendTo()
Sleep for a while ... 
05-09-2017 16:28:44.677139 {DBG} "IqrfSpiChannel.cpp" ln:177 sendTo()
Trying to sent: 1.4
05-09-2017 16:28:44.678231 {DBG} "IqrfSpiChannel.cpp" ln:197 sendTo()
status.isDataReady="0x0" status.dataNotReadyStatus="0x0" 
05-09-2017 16:28:44.678322 {DBG} "IqrfSpiChannel.cpp" ln:201 sendTo()
Sleep for a while ... 
05-09-2017 16:28:44.778504 {DBG} "DpaHandler.cpp" ln:71 SendDpaMessage()
Setting user timeout: m_defaultTimeoutMs="400" 
05-09-2017 16:28:44.778612 {DBG} "DpaTransfer.cpp" ln:47 ProcessSentMessage()
{ENTER} 
05-09-2017 16:28:44.778707 {DBG} "DpaTransfer.cpp" ln:294 SetTimingForCurrentTransfer()
{ENTER} 
05-09-2017 16:28:44.778795 {DBG} "DpaTransfer.cpp" ln:321 SetTimingForCurrentTransfer()
Expected duration to wait :m_expectedDurationMs="400" 
05-09-2017 16:28:44.778880 {INF} "DpaTransfer.cpp" ln:326 SetTimingForCurrentTransfer()
Transfer status: started
05-09-2017 16:28:44.778964 {DBG} "DpaTransfer.cpp" ln:328 SetTimingForCurrentTransfer()
{LEAVE} 
05-09-2017 16:28:44.779047 {DBG} "DpaTransfer.cpp" ln:70 ProcessSentMessage()
{LEAVE} 
05-09-2017 16:28:44.779130 {DBG} "DpaHandler.cpp" ln:80 SendDpaMessage()
{LEAVE} 
05-09-2017 16:28:44.779220 {DBG} "DpaTransfer.cpp" ln:366 CheckTimeout()
Time to wait: remains="400" 
05-09-2017 16:28:44.779363 {DBG} "DpaHandler.cpp" ln:229 ExecuteDpaTransaction()
Conditional wait - time to wait yet: remains="400" 
05-09-2017 16:28:44.956522 {DBG} "DpaHandler.cpp" ln:104 ResponseMessageHandler()
>>>>>>>>>>>>>>>>>>
Received from IQRF interface: 
00 00 ff 3f 00 00 80 00 01 03 00 fd 26 00 00 00 ...?........&...
00 00 00 02                                     ....            

05-09-2017 16:28:44.956655 {DBG} "DpaTransfer.cpp" ln:75 ProcessReceivedMessage()
{ENTER} 
05-09-2017 16:28:44.957044 {WAR} "DpaHandler.cpp" ln:139 ProcessMessage()
Caught Process received message error...: std::logic_error: Response is expected.
05-09-2017 16:28:44.957169 {ERR} "DpaHandler.cpp" ln:119 ResponseMessageHandler()
Received message has not been processed!
05-09-2017 16:28:44.957288 {INF} "DpaHandler.cpp" ln:126 ResponseMessageHandler()
Notify from ResponseMessageHandler: message received.
05-09-2017 16:28:44.957421 {DBG} "DpaTransfer.cpp" ln:366 CheckTimeout()
Time to wait: remains="222" 
05-09-2017 16:28:44.957511 {DBG} "DpaHandler.cpp" ln:229 ExecuteDpaTransaction()
Conditional wait - time to wait yet: remains="222" 
05-09-2017 16:28:45.179744 {DBG} "DpaTransfer.cpp" ln:366 CheckTimeout()
Time to wait: remains="0" 
05-09-2017 16:28:45.180033 {DBG} "DpaTransfer.cpp" ln:366 CheckTimeout()
Time to wait: remains="-1" 
05-09-2017 16:28:45.180124 {INF} "DpaTransfer.cpp" ln:383 CheckTimeout()
Transfer status: timeout
05-09-2017 16:28:45.180209 {DBG} "DpaTransfer.cpp" ln:332 ProcessStatus()
{ENTER} 
05-09-2017 16:28:45.180297 {DBG} "DpaTransfer.cpp" ln:366 CheckTimeout()
Time to wait: remains="-1" 
05-09-2017 16:28:45.180379 {DBG} "DpaTransfer.cpp" ln:341 ProcessStatus()
{LEAVE} 
05-09-2017 16:28:45.180627 {WAR} "DaemonController.cpp" ln:390 startDpa()
Throwing std::logic_error: Cannot get TR parameters
05-09-2017 16:28:45.180858 {ERR} "DaemonController.cpp" ln:403 startDpa()
There was an error during DPA handler creation: Cannot get TR parameters
...

spinarr, 2017-09-05:
It also looks that DPA req getting module info has not been received by the 
DCTR module since SPI channel is not in ready state yet.

fmikulu, 2017-09-09:
We shall clarify StartUp in general. The very first DpaRequest is getting info. No DCTR reset is sent. It may be side effect if all starts together? How to cope with it? Waiting for Async DpaResponse after start may be a problem in case of Hot Start (just daemon - coordinator running) and Could Start - everything from scratch.

I propose to postpone it  till solving #63 or solve it together. Another possibility is to repeat DpaRequest Info in case of failure. 

DpaTransaction shall be ready to be interrupted by Async DpaResponse (handled outside pending transaction) and continue - ordinary DpaResponse may come later.

spinarr, 2017-09-11:
Tests also revealed that Getting module info is sent too early, SPI channel is not ready,
so at least to wait for SPI channel to be ready (0x80) - this may be also related to 
handling in cutils.

Ok, we do not need to wait for async message, or wait with set timeout before we
implement handling async msg while dpa transaction, transfer in progress.

fmikulu, 2017-09-15:
Getting Module Info shall be done when IQRF interface ready, see https://github.com/iqrfsdk/cutils/issues/5

fmikulu, 2017-09-21:
Getting Module Info is postponed now when interface is ready. It still may interfere with AsyncDpaMessage if TR is started simultaneously with the daemon. It will be solved in scope of https://github.com/iqrfsdk/clibdpa/issues/17

closed 2017-09-24 by spinarr