iqrfsdk/iqrf-daemon #68: IQRFAPP conf - anything wrong?

spinarr, 2017-09-07:
branch: https://github.com/iqrfsdk/iqrf-daemon/tree/feature/link-clibdpa-review-insteps

07-09-2017 19:34:04.58607 {DBG} "MqMessaging.cpp" ln:94 handleMessageFromMq()
==================================
Received from MQ: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 66 6f 72 77 61 72 64 69 6e 67 22 md":"forwarding"
7d 20                                           }.              

07-09-2017 19:34:04.58824 {DBG} "BaseService.cpp" ln:87 handleMsgFromMessaging()
==================================
Received from MESSAGING: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 66 6f 72 77 61 72 64 69 6e 67 22 md":"forwarding"
7d 20                                           }.              

07-09-2017 19:34:04.58911 {WAR} "ObjectFactory.h" ln:54 createObject()
Throwing std::logic_error: Unregistered parser for: id="{"ctype":"conf","type":"mode","cmd":"forwarding"}" 
07-09-2017 19:34:04.59066 {DBG} "DaemonController.cpp" ln:236 setMode()
{ENTER} mode="2" 
07-09-2017 19:34:04.59114 {DBG} "UdpMessaging.cpp" ln:70 resetExclusive()
{ENTER} 
07-09-2017 19:34:04.59195 {DBG} "UdpMessaging.cpp" ln:72 resetExclusive()
{LEAVE}

sudo iqrfapp "{\"ctype\":\"conf\",\"type\":\"mode\",\"cmd\":\"forwarding\"}"

after that no reaction on any other similar iqrfapp cmd.

fmikulu, 2017-09-07:
Throwing exception shall be OK as it tries all registered parsers.

If I understand well:
The very first:
`sudo iqrfapp "{"ctype":"conf","type":"mode","cmd":"forwarding"}"`
Works, but no other:
`sudo iqrfapp "{"ctype":"conf","type":"mode","cmd":"operational"}"`
`sudo iqrfapp "{"ctype":"conf","type":"mode","cmd":"service"}"`
`sudo iqrfapp "{"ctype":"conf","type":"mode","cmd":"forwarding"}"`
Is accepted?

Can you attach whole log file or at lest the excerpt around untreated iqrfapp commands?

spinarr, 2017-09-08:
it looks like following:

pi@raspberrypi:~/IQRF/pkg/iqrf-daemon/scripts $ ./mq-iqrfapp-switchmode.sh service
Sending conf request to change gw mode to service
cmd="service"  done

**OK**

pi@raspberrypi:~/IQRF/pkg/iqrf-daemon/scripts $ ./mq-iqrfapp-switchmode.sh forwarding
Sending conf request to change gw mode to forwarding

**NO OUTPUT, but mode set, see attached log**

pi@raspberrypi:~/IQRF/pkg/iqrf-daemon/scripts $ ./mq-iqrfapp-switchmode.sh operational
Sending conf request to change gw mode to operational

**NO OUTPUT, no reaction in daemon**

08-09-2017 14:24:16.164917 {DBG} "MqMessaging.cpp" ln:94 handleMessageFromMq()
==================================
Received from MQ: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 73 65 72 76 69 63 65 22 7d 20    md":"service"}. 

08-09-2017 14:24:16.165261 {DBG} "BaseService.cpp" ln:87 handleMsgFromMessaging()
==================================
Received from MESSAGING: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 73 65 72 76 69 63 65 22 7d 20    md":"service"}. 

08-09-2017 14:24:16.165485 {WAR} "ObjectFactory.h" ln:54 createObject()
Throwing std::logic_error: Unregistered parser for: id="{"ctype":"conf","type":"mode","cmd":"service"}" 
08-09-2017 14:24:16.165918 {DBG} "DaemonController.cpp" ln:236 setMode()
{ENTER} mode="1" 
08-09-2017 14:24:16.166037 {DBG} "DaemonController.cpp" ln:691 stopDpa()
{ENTER} 
08-09-2017 14:24:16.166154 {DBG} "DaemonController.cpp" ln:692 stopDpa()
Try to destroy: m_dpaHandler="0x1588828" 
08-09-2017 14:24:16.166318 {DBG} "DaemonController.cpp" ln:695 stopDpa()
{LEAVE} 
08-09-2017 14:24:16.166413 {DBG} "UdpMessaging.cpp" ln:58 setExclusive()
{ENTER} 
08-09-2017 14:24:16.166514 {DBG} "UdpMessaging.cpp" ln:65 setExclusive()
{LEAVE} 
08-09-2017 14:24:16.166609 {INF} "DaemonController.cpp" ln:277 setMode()
Set mode service
08-09-2017 14:24:16.166701 {DBG} "DaemonController.cpp" ln:288 setMode()
{LEAVE} 
08-09-2017 14:24:16.166890 {DBG} "MqMessaging.cpp" ln:87 sendMessage()
63 6d 64 3d 22 73 65 72 76 69 63 65 22 20 20 64 cmd="service"..d
6f 6e 65                                        one             

08-09-2017 14:24:16.167166 {DBG} "MqChannel.cpp" ln:238 sendTo()
Send to MQ: 
63 6d 64 3d 22 73 65 72 76 69 63 65 22 20 20 64 cmd="service"..d
6f 6e 65                                        one             

08-09-2017 14:24:16.167322 {DBG} "MqChannel.cpp" ln:230 connect()
openMqWrite() opened: m_remoteMqName="/iqrf-daemon-100" 
08-09-2017 14:24:49.523309 {DBG} "MqMessaging.cpp" ln:94 handleMessageFromMq()
==================================
Received from MQ: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 66 6f 72 77 61 72 64 69 6e 67 22 md":"forwarding"
7d 20                                           }.              

08-09-2017 14:24:49.523651 {DBG} "BaseService.cpp" ln:87 handleMsgFromMessaging()
==================================
Received from MESSAGING: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 66 6f 72 77 61 72 64 69 6e 67 22 md":"forwarding"
7d 20                                           }.              

08-09-2017 14:24:49.523828 {WAR} "ObjectFactory.h" ln:54 createObject()
Throwing std::logic_error: Unregistered parser for: id="{"ctype":"conf","type":"mode","cmd":"forwarding"}" 
08-09-2017 14:24:49.524205 {DBG} "DaemonController.cpp" ln:236 setMode()
{ENTER} mode="2" 
08-09-2017 14:24:49.524320 {DBG} "UdpMessaging.cpp" ln:70 resetExclusive()
{ENTER} 
08-09-2017 14:24:49.524420 {DBG} "UdpMessaging.cpp" ln:72 resetExclusive()
{LEAVE} 
08-09-2017 14:24:49.524521 {DBG} "DpaHandler.cpp" ln:31 DpaHandler()
Ctor default user timeout: m_defaultTimeoutMs="200" 
08-09-2017 14:24:49.524634 {DBG} "DpaTransfer.cpp" ln:348 ProcessStatus()
{ENTER} 
08-09-2017 14:24:49.524729 {INF} "DpaTransfer.cpp" ln:363 CheckTimeout()
Transfer status: created
08-09-2017 14:24:49.524821 {DBG} "DpaTransfer.cpp" ln:354 ProcessStatus()
{LEAVE}

spinarr, 2017-09-08:
It looks like that after that even MQTT channel stoped working ...

spinarr, 2017-09-08:
It looks like that something breaks after switching from service mode, regardless MQ/MQTT channel and operational/forwarding mode.

fmikulu, 2017-09-17:
@spinarr please retest with latest iqrfapp

spinarr, 2017-09-18:
It breaks even via MQTT, still breaks when switching from service to other modes.

fmikulu, 2017-09-21:
Fixed dd88855ecf59b2c3a9cfd7656b6b684a24f18ac2
@spinarr please verify

spinarr, 2017-09-21:
It is working now, however, there is trouble to get tr module info when switching from service mode

21-09-2017 19:24:54.76989 {DBG} "BaseService.cpp" ln:96 handleMsgFromMessaging()
==================================
Received from MESSAGING: 
7b 22 63 74 79 70 65 22 3a 22 63 6f 6e 66 22 2c {"ctype":"conf",
22 74 79 70 65 22 3a 22 6d 6f 64 65 22 2c 22 63 "type":"mode","c
6d 64 22 3a 22 66 6f 72 77 61 72 64 69 6e 67 22 md":"forwarding"
7d 20                                           }.              

21-09-2017 19:24:54.77141 {DBG} "DaemonController.cpp" ln:242 setMode()
{ENTER} mode="2" 
21-09-2017 19:24:54.77223 {DBG} "UdpMessaging.cpp" ln:70 resetExclusive()
{ENTER} 
21-09-2017 19:24:54.77330 {DBG} "UdpMessaging.cpp" ln:72 resetExclusive()
{LEAVE} 
21-09-2017 19:24:54.77468 {DBG} "DpaHandler.cpp" ln:31 DpaHandler()
Ctor default user timeout: m_defaultTimeoutMs="200" 
21-09-2017 19:24:54.77642 {DBG} "DpaTransfer.cpp" ln:348 ProcessStatus()
{ENTER} 
21-09-2017 19:24:54.77795 {INF} "DpaTransfer.cpp" ln:363 CheckTimeout()
Transfer status: created
21-09-2017 19:24:54.77897 {DBG} "DpaTransfer.cpp" ln:354 ProcessStatus()
{LEAVE} 
21-09-2017 19:24:54.78068 {DBG} "DaemonController.cpp" ln:102 executeDpaTransactionFunc()
Dpa interface is in exclusiveMode
21-09-2017 19:24:54.78181 {WAR} "DaemonController.cpp" ln:421 startDpa()
Throwing std::logic_error: Cannot get TR parameters
21-09-2017 19:24:54.78318 {ERR} "DaemonController.cpp" ln:434 startDpa()
There was an error during DPA handler creation: Cannot get TR parameters
21-09-2017 19:24:54.78371 {INF} "DaemonController.cpp" ln:268 setMode()
Set mode forwarding
21-09-2017 19:24:54.78447 {DBG} "DaemonController.cpp" ln:294 setMode()
{LEAVE} 
21-09-2017 19:24:54.78549 {DBG} "MqMessaging.cpp" ln:87 sendMessage()
63 6d 64 3d 22 66 6f 72 77 61 72 64 69 6e 67 22 cmd="forwarding"
20 20 64 6f 6e 65                               ..done          

21-09-2017 19:24:54.78676 {DBG} "MqChannel.cpp" ln:301 sendTo()
Send to MQ: 
63 6d 64 3d 22 66 6f 72 77 61 72 64 69 6e 67 22 cmd="forwarding"
20 20 64 6f 6e 65                               ..done          

21-09-2017 19:24:54.78762 {DBG} "MqChannel.cpp" ln:121 writeMq()
{ENTER} toWrite="22" 
21-09-2017 19:24:54.79743 {DBG} "MqChannel.cpp" ln:127 writeMq()
{LEAVE} retval="1"

fmikulu, 2017-09-21:
Throwing std::logic_error: Cannot get TR parameters solved by 36aab0c7b54bb563fa42453d0c9c130d1aa8c135

spinarr, 2017-09-21:
now is fine, very good

closed 2017-09-21 by spinarr