iqrfsdk/iqrf-daemon #70: Review current implementation of ctype:conf switching of gw mode

spinarr, 2017-09-14:
Implement ctype:conf according to the definition
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructureConf-v1

fmikulu, 2017-09-15:
For runtime cfg. update we have only support for **mode** switching (operational/forwarding/service). It make sense to review/update implementation of JSON **mode** msg according the definition.

Does it make sense to develop in scope V1 configuring Trace component?
- level
- file name
- file size

If yes lets make dedicated issue

spinarr, 2017-09-18:
Only review JSON mode msg:

cmd=operational/forwarding/service

input: "{\"ctype\":\"conf\",\"type\":\"mode\",\"cmd\":\"$cmd\"}"
output: cmd="operational" done

fmikulu, 2017-09-22:
Implemented b05cfe43b3efa362aba75c789fef1b3f8f3386c3
In feature/ConfParser
@spinarr please try

spinarr, 2017-09-24:
Now it follows: 

https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructureConf-v1
Gateway runtime parameters
There is a mode in which gateway operates.

It is ok.

closed 2017-09-24 by spinarr