iqrfsdk/iqrf-daemon #74: Missing systemd service

mienkofax, 2017-09-21:
When update.sh script is running, it is not possible to install a service because systemd service is missing - it was deleted in this commit [Iqrfapp](https://github.com/iqrfsdk/iqrf-daemon/commit/86a96c690da4a45f7cfdcb0b0c48e973f11d21f6#diff-3753c4c43ad751fe2980bb8916810cee)

spinarr, 2017-09-21:
Why do you need to install service?
Service is installed when you install daemon from repository ... apt-get install iqrf-daemon.

mienkofax, 2017-09-21:
My idea about installation of whole application follows:

`git clone ...`
`cd daemon`
`mkdir build`
`cmake ..`
`sudo make install //install iqrf_startup to /usr/bin/iqrf_startup and copy config file to /etc/iqrf-daemon/`

But following libraries are necessary: paho, libspi, clibcdc, cutils, clibdpa. It is not problem, I download repositories with libraries and I use usual commands:

 `git clone, cd .., mkdir build, cmake .., sudo make install....`

But no library can be installed, installations are in comment everywhere. I try to set paths, all is compilled. But I think that this procedure is very unhappy. Particular libraries should be found using cmake in a standard position /usr/lib (/usr/lib, /usr/local/lib, ...)

The last part is standard installation using `sudo make install` but this also does not work. In this installation I expected installing of bin + configuration files + services (The same what is included in a deb packet).

Take it easy, I can create and send patch for daemon and for particular libraries. But it is possible that I overlooked something and this installation makes sense.

spinarr, 2017-09-24:
I see, patch is much appreciated, thanks a lot.

spinarr, 2017-10-10:
Just checking if there is any progress or anything that can be done,

spinarr, 2018-02-27:
Python and Make:
https://github.com/iqrfsdk/iqrf-daemon/tree/master/scripts/py

closed 2018-03-01 by spinarr