iqrfsdk/iqrf-daemon #75: Core dump after upgrade to v8.0

mienkofax, 2017-09-26:
After the upgrade to v8.0 the daemon started with core dump and stopped working.

```
systemctl restart iqrf-daemon.service

Stopped IQRF daemon iqrf_startup.
pc systemd[1]: Started IQRF daemon iqrf_startup.
pc iqrf_startup[15731]: /usr/bin/iqrf_startup started
pc iqrf_startup[15731]: terminate called after throwing an instance of 'std::logic_error'
pc iqrf_startup[15731]:   what():  Unexpected configuration: cfgVersion="v0.8" expected: v0.0
pc iqrf_startup[15731]: Loading configuration file: cfgFileName="/etc/iqrf-daemon/config.json"
pc systemd[1]: iqrf-daemon.service: Main process exited, code=killed, status=6/ABRT
pc systemd[1]: iqrf-daemon.service: Unit entered failed state.
pc systemd[1]: iqrf-daemon.service: Failed with result 'signal'.
```

spinarr, 2017-09-26:
What is target system?

mienkofax, 2017-09-26:
target system: Linux pc 4.4.0-96-generic #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

I tried install iqrd daemon in clean system inside Virtual-box (Ubuntu x64) and the same problem.

Roman3349, 2017-09-26:
I saw it in package for Ubuntu 16.04.

```json
{
  "Configuration": "v0.8",
  "ConfigurationDir": "/etc/iqrf-daemon",
  "WatchDogTimeoutMilis": 10000,
  "Mode": "operational",

  "Components": [
    {
      "ComponentName": "BaseService",
      "Enabled": true
    },
    {
      "ComponentName": "TracerFile",
      "Enabled":  true
    },
    {
      "ComponentName": "IqrfInterface",
      "Enabled":  true
    },
    {
      "ComponentName": "UdpMessaging",
      "Enabled":  true
    },
    {
      "ComponentName": "MqttMessaging",
      "Enabled":  true
    },
    {
      "ComponentName": "MqMessaging",
      "Enabled":  true
    },
    {
      "ComponentName": "Scheduler",
      "Enabled":  true
    },
    {
      "ComponentName": "SimpleSerializer",
      "Enabled":  true
    },
    {
      "ComponentName": "JsonSerializer",
      "Enabled":  true
    }
  ]
}
```

fmikulu, 2017-09-26:
There should be as expected: v0.0
It is not version of cfg files, not version of release.

spinarr, 2017-09-26:
my mistake, have to fix it

fmikulu, 2017-09-26:
Actually, It isn't too much intuitive. Maybe we can omit it completely and replace by soming else in V2

spinarr, 2017-10-01:
Will we keep "Configuration": "v0.0" for release v1?

fmikulu, 2017-10-01:
We can change it to some nicer value. However it is not mention as release version, but rather configuration structure version, so the same structure can survive more releases.  It makes sense if we want to support backward compatibility of configuration files. So lets start with v1.0 for release v1 and we will see ...

closed 2017-10-01 by spinarr