iqrfsdk/iqrf-daemon #78: Confirm 2 instances of the daemon running at the same time

spinarr, 2017-10-05:
Make sure we can run 2 instances of the daemon at the same time
https://github.com/iqrfsdk/iot-starter-kit/tree/master/workshop/iqrf-docker

fmikulu, 2017-10-05:
# Running two instances:
I cannot test it right now as I haven't configured two running IQRF meshes.
However it is possible if the two instances didn't have conflicting configurations (with respect to system sources).  e.g;

**./iqrf-startup ./configuration1/config.json**
**./iqrf-startup ./configuration2/config.json**

These configuration files have to be considered (if we need these components enabled):
## IqrfInterface.json
(https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/IqrfInterface.json)
There must be distinguished interfaces, e.g.
```json
{
"IqrfInterface": "/dev/ttyACM0"
}
```
and

```json
{
"IqrfInterface": "/dev/spidev0.0"
}
```
## MqMessaging.json
(https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/MqMessaging.json)
There must be distinguished MQ channel, e.g.
```json
"Properties": { 
    "LocalMqName": "iqrf-daemon-110",
    "RemoteMqName": "iqrf-daemon-100"
}
```
and

```json
"Properties": {
    "LocalMqName": "iqrf-daemon-210",
    "RemoteMqName": "iqrf-daemon-200"
}
```
Don't forget to set counterpart channels for `iqrfapp` 
https://github.com/iqrfsdk/iqrf-daemon/blob/master/applications/iqrfapp/iqrfapp.json
`iqrfapp` has default values, however it is configurable as can be seen from its source code:
```cpp
  std::string m_cfgFileName = "iqrfapp.json";
  std::string localMqName = "iqrf-daemon-100";
  std::string remoteMqName = "iqrf-daemon-110";

  std::ifstream f;

  //open in working directory
  f.open(m_cfgFileName);

  if (!f.is_open()) {
    std::string pth = "/etc/iqrf-daemon/";
    m_cfgFileName = pth + m_cfgFileName;

    //open in default path
    f.open(m_cfgFileName);
  }
```
## TracerFile.json
https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/TracerFile.json
Just assure different file names (if emty - stdout is used)
```json
{
 "TraceFileName": "trc1"
}
```
and
```json
{
 "TraceFileName": "trc2"
}
```
## UdpMessaging.json
https://github.com/iqrfsdk/iqrf-daemon/blob/master/daemon/iqrf_startup/configuration/UdpMessaging.json
Use different ports:
```json
"Properties": {
     "RemotePort": 55000,
     "LocalPort": 55300
}
```
and

```json
"Properties": {
     "RemotePort": 55001,
     "LocalPort": 55301
}
```
In this case it is necessary to configure IDE4 accordingly

vojta-novak, 2017-10-07:
Hi, 
this is a good point.
I would just add a few ideas.
1) The port does not need to be changed, it can be remapped when loading the container.
example: container 1 (-p 80:80), container 2 (-p 81:80),
2) It would be good to determine the behavior if the setting is the same. Sometimes it would be interesting for two containers to have the same settings.

spinarr, 2017-10-16:
Confirmed.

https://github.com/iqrfsdk/iqrf-daemon/tree/master/docker
...
Ready for 2 IQRF coordinators/networks on single GW
...

closed 2017-10-16 by spinarr