iqrfsdk/iqrf-daemon #79: Review RAW and RAW-HDP serializers

spinarr, 2017-10-25:
Before release:

- check timestamp format, suggestion:
http://support.sas.com/documentation/cdl/en/lrdict/64316/HTML/default/viewer.htm#a003169814.htm

- check hwpid in json response for raw-hdp
(should be present)

- leave only "." as separator among bytes

Wiki page shall be also revieved.
https://github.com/iqrfsdk/iqrf-daemon/wiki/JsonStructureDpa-v1

fmikulu, 2017-10-28:
Regarding timestamp:

We have:
**yyyy-mm-dd hh:mm:ss.ffffff**

As specified by http://support.sas.com/documentation/cdl/en/lrdict/64316/HTML/default/viewer.htm#a003169814.htm

Datetime | yyyy-mm-ddThh:mm:ss.ffffff | 2008-09-15T15:53:00 | E8601DTw.d
-- | -- | -- | --

Just missing "T" is required to be there?

Roman3349, 2017-10-28:
I hope that the correct implementation of ISO 8601 in timestamps is in this commit: https://github.com/iqrfsdk/iqrf-daemon/commit/e5058a475b7f2217f1ba1b08fd39c3ceb68a35ee.

fmikulu, 2017-11-05:
- Missing **T** required by ISO 8601 fixed by @Roman3349  merged via: 516a6e928ac6a15bda5050f5a6cc360bb4daa22c

- Space as the separator in binary format is still accepted, however '.' is used in responses.

- HWPID is not mandatory field defaulted to **ffff** and thus not present in JsonDpaResponse. @spinarr : Shall it be mandatory regardless passed or not in counterpart  JsonDpaRequest?

- I believe there are lowecase in every hexa coding used in JsonSerializer as it is default option for std::stream. @spinarr : Can you clarify where it is not the case - maybe just some wiki example?

spinarr, 2017-11-06:
- hwpid should be mandatory regardless 
(0xffff is not mirrored in dpa response ... there is 0x0000 or custom hwpid)

- lowercase is default and all examples are with lowercase 
(there was request to define/fix only lowercase)

- shall res_data be renamed to rdata as in the wiki?

spinarr, 2017-11-08:
- also have req_data in the request
- req_data and res_data could be rdata

closed 2017-12-15 by spinarr