iqrfsdk/iqrf-daemon #80: Extending IqrfInterface

spinarr, 2017-10-25:
For SPI:
https://github.com/iqrfsdk/clibspi

- allow to define mapping of these pins via IqrfInterface.json
https://github.com/iqrfsdk/clibspi/blob/master/include/machines_def.h

fmikulu, 2017-11-05:
**I recommend to postpone** this feature as it could significantly delay **V1.0.0** iqrf-daemon delivery. Lets do it in scope of some **V1.0.X**

The most of the feature has to be implemented in clibspi: https://github.com/iqrfsdk/clibspi/issues/5

Meantime there is a parallel work in branch: https://github.com/iqrfsdk/clibspi/tree/upload and it would mean a lot of merge conflicts in future.

Anyway is it secure at all from HW point of view just to change IO pins from upper layers? I believe we would use shared library as expected in V2. It this case it means just to compile a special clibspi for a special HW with hardcoded IO pins as the pins are hard-wired in fact.

For iqrf-daemon it seems reasonable to split IqrfInterface component and provide separate configurations for both SPI and CDC interfaces as it is misleading to mix dedicated parameters for SPI and CDC. Note existing implementation of iqrf-daemon just pre-parses *"IqrfInterface"* item from *IqrfInterface.json* file and instantiate interface accordingly:

```cpp
      size_t found = m_iqrfInterfaceName.find("spi");
      if (found != std::string::npos)
        m_iqrfInterface = ant_new IqrfSpiChannel(m_iqrfInterfaceName);
      else
        m_iqrfInterface = ant_new IqrfCdcChannel(m_iqrfInterfaceName);
```

spinarr, 2017-11-06:
- yes, we postpone for 1.X.0

- let see about shared spi library in v2
(who would maintain special builds ... hw producer?)

- we shall provide addtional SpiInterface.json and CdcInterface.json for v1 in order to be able
to add additional config parameters separately later on? We could also keep single IqrfInterface.json

...
"interface":"spi"
...
"spi":{
...
}

"cdc":{
...
}

spinarr, 2018-02-22:
Enabling functionality is here:
https://github.com/iqrfsdk/clibspi/tree/develop

More specifically:
SPI_IQRF_DECLSPEC int spi_iqrf_initAdvanced(spi_iqrf_config_struct *configStruct);

Example:
https://github.com/iqrfsdk/clibspi/blob/develop/examples/spi/spi_example_dpa/spi_example_dpa.c#L165

Please, @fmikulu let us agree on integration (configuration IqrfInterface.json) to the daemon.

fmikulu, 2018-02-24:
Implemented via 022c8213010126cfb834945715b75cfd8e3bed3f

It is necessary to get the latest master branch of clibspi, cutils, clibdpa

@spinarr please verify (works on my RPI)

fmikulu, 2018-02-24:
I quess it solves #92 as well?

closed 2018-03-12 by spinarr