iqrfsdk/iqrf-daemon #81: mqtt connect and error_timeout

spinarr, 2017-10-27:
broker = iot.eclipse.org:1883

log:
27-10-2017 03:07:03.894232 {WAR} "MqttMessaging.cpp" ln:442 connlost()
Connection lost: cause="nullptr"
27-10-2017 03:07:03.894376 {DBG} "MqttMessaging.cpp" ln:315 connect()
{ENTER}
27-10-2017 03:07:03.894529 {DBG} "MqttMessaging.cpp" ln:326 connect()
{LEAVE}
27-10-2017 03:07:03.894744 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:03.894948 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="1"
27-10-2017 03:07:04.895344 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:05.45212 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="2"
27-10-2017 03:07:05.45703 {INF} "MqttMessaging.cpp" ln:335 onConnect()
Connect succeded: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:05.45856 {DBG} "MqttMessaging.cpp" ln:344 onConnect()
Subscribing: m_mqttTopicRequest="Iqrf/DpaRequest" m_mqttQos="1"
27-10-2017 03:07:05.195213 {INF} "MqttMessaging.cpp" ln:373 onSubscribe()
Subscribe succeded: m_mqttTopicRequest="Iqrf/DpaRequest" m_mqttQos="1"
27-10-2017 03:07:05.383981 {DBG} "MqttMessaging.cpp" ln:407 msgarrvd()
topic="Iqrf/DpaRequest"
27-10-2017 03:07:05.384203 {DBG} "MqttMessaging.cpp" ln:248 handleMessageFromMqtt()

Client receives:
https://www.eclipse.org/paho/clients/js/utility/

Iqrf/DpaResponse | { "ctype": "dpa", "type": "raw-hdp", "msgid": "1", "nadr": "01", "pnum": "ff", "pcmd": "3f", "res_data": "01 03 01 fe 26 00 00 02 00 00 00 01 00 00 00 00 00 00 00 40", "hwpid": "ffff", "rcode": "00", "dpaval": "67", "response": "01 00 ff bf 02 00 00 67 01 03 01 fe 26 00 00 02 00 00 00 01 00 00 00 00 00 00 00 40", "status": "STATUS_NO_ERROR" }

it is ok, but often we get:

Iqrf/DpaResponse | { "ctype": "dpa", "type": "raw-hdp", "msgid": "1", "nadr": "01", "pnum": "ff", "pcmd": "3f", "hwpid": "ffff", "rcode": "00", "dpaval": "00", "response": "", "status": "ERROR_TIMEOUT" }

I do not see any timeout regarding DPA transfer in the log, but still there is timeout.
Must be connected to the broken MQTT connection.

27-10-2017 03:07:08.244827 {DBG} "MqttMessaging.cpp" ln:258 sendTo()
Sending to MQTT: topic="Iqrf/DpaResponse"
7b 0a 20 20 20 20 22 63 74 79 70 65 22 3a 20 22 {....."ctype":."
64 70 61 22 2c 0a 20 20 20 20 22 74 79 70 65 22 dpa",....."type"
3a 20 22 72 61 77 2d 68 64 70 22 2c 0a 20 20 20 :."raw-hdp",....
20 22 6d 73 67 69 64 22 3a 20 22 31 22 2c 0a 20 ."msgid":."1",..
20 20 20 22 6e 61 64 72 22 3a 20 22 30 31 22 2c ..."nadr":."01",
0a 20 20 20 20 22 70 6e 75 6d 22 3a 20 22 66 66 ....."pnum":."ff
22 2c 0a 20 20 20 20 22 70 63 6d 64 22 3a 20 22 ",....."pcmd":."
33 66 22 2c 0a 20 20 20 20 22 72 65 73 5f 64 61 3f",....."res_da
74 61 22 3a 20 22 30 31 20 30 33 20 30 31 20 66 ta":."01.03.01.f
65 20 32 36 20 30 30 20 30 30 20 30 32 20 30 30 e.26.00.00.02.00
20 30 30 20 30 30 20 30 31 20 30 30 20 30 30 20 .00.00.01.00.00.
30 30 20 30 30 20 30 30 20 30 30 20 30 30 20 34 00.00.00.00.00.4
30 22 2c 0a 20 20 20 20 22 68 77 70 69 64 22 3a 0",....."hwpid":
20 22 66 66 66 66 22 2c 0a 20 20 20 20 22 72 63 ."ffff",....."rc
6f 64 65 22 3a 20 22 30 30 22 2c 0a 20 20 20 20 ode":."00",.....
22 64 70 61 76 61 6c 22 3a 20 22 36 37 22 2c 0a "dpaval":."67",.
20 20 20 20 22 72 65 73 70 6f 6e 73 65 22 3a 20 ...."response":.
22 30 31 20 30 30 20 66 66 20 62 66 20 30 32 20 "01.00.ff.bf.02.
30 30 20 30 30 20 36 37 20 30 31 20 30 33 20 30 00.00.67.01.03.0
31 20 66 65 20 32 36 20 30 30 20 30 30 20 30 32 1.fe.26.00.00.02
20 30 30 20 30 30 20 30 30 20 30 31 20 30 30 20 .00.00.00.01.00.
30 30 20 30 30 20 30 30 20 30 30 20 30 30 20 30 00.00.00.00.00.0
30 20 34 30 22 2c 0a 20 20 20 20 22 73 74 61 74 0.40",....."stat
75 73 22 3a 20 22 53 54 41 54 55 53 5f 4e 4f 5f us":."STATUS_NO_
45 52 52 4f 52 22 0a 7d                         ERROR".}

27-10-2017 03:07:08.245345 {WAR} "MqttMessaging.cpp" ln:442 connlost()
Connection lost: cause="nullptr"
27-10-2017 03:07:08.245426 {DBG} "MqttMessaging.cpp" ln:315 connect()
{ENTER}
27-10-2017 03:07:08.245553 {DBG} "MqttMessaging.cpp" ln:326 connect()
{LEAVE}
27-10-2017 03:07:08.245810 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:08.245987 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="1"
27-10-2017 03:07:09.246385 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:09.246591 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="2"
27-10-2017 03:07:09.396164 {WAR} "MqttMessaging.cpp" ln:433 onSendFailure()
Message sent failure: response->code="-11"
27-10-2017 03:07:09.396360 {INF} "MqttMessaging.cpp" ln:335 onConnect()
Connect succeded: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
27-10-2017 03:07:09.396434 {DBG} "MqttMessaging.cpp" ln:344 onConnect()
Subscribing: m_mqttTopicRequest="Iqrf/DpaRequest" m_mqttQos="1"
27-10-2017 03:07:09.546268 {INF} "MqttMessaging.cpp" ln:373 onSubscribe()
Subscribe succeded: m_mqttTopicRequest="Iqrf/DpaRequest" m_mqttQos="1"
27-10-2017 03:07:09.735920 {DBG} "MqttMessaging.cpp" ln:407 msgarrvd()
topic="Iqrf/DpaRequest"
27-10-2017 03:07:09.736147 {DBG} "MqttMessaging.cpp" ln:248 handleMessageFromMqtt()
==================================
Received from MQTT:
7b 0d 0a 20 20 22 63 74 79 70 65 22 3a 20 22 64 {...."ctype":."d
70 61 22 2c 0d 0a 20 20 22 74 79 70 65 22 3a 20 pa",...."type":.
22 72 61 77 2d 68 64 70 22 2c 0d 0a 20 20 22 6d "raw-hdp",...."m
73 67 69 64 22 3a 20 22 31 22 2c 0d 0a 20 20 22 sgid":."1",...."
6e 61 64 72 22 3a 20 22 30 31 22 2c 0d 0a 20 20 nadr":."01",....
22 70 6e 75 6d 22 3a 20 22 66 66 22 2c 0d 0a 20 "pnum":."ff",...
20 22 70 63 6d 64 22 3a 20 22 33 66 22 2c 0d 0a ."pcmd":."3f",..
20 20 22 68 77 70 69 64 22 3a 20 22 66 66 66 66 .."hwpid":."ffff
22 0d 0a 7d                                     "..}

27-10-2017 03:07:09.736285 {DBG} "BaseService.cpp" ln:96 handleMsgFromMessaging()
==================================
Received from MESSAGING:
7b 0d 0a 20 20 22 63 74 79 70 65 22 3a 20 22 64 {...."ctype":."d
70 61 22 2c 0d 0a 20 20 22 74 79 70 65 22 3a 20 pa",...."type":.
22 72 61 77 2d 68 64 70 22 2c 0d 0a 20 20 22 6d "raw-hdp",...."m
73 67 69 64 22 3a 20 22 31 22 2c 0d 0a 20 20 22 sgid":."1",...."
6e 61 64 72 22 3a 20 22 30 31 22 2c 0d 0a 20 20 nadr":."01",....
22 70 6e 75 6d 22 3a 20 22 66 66 22 2c 0d 0a 20 "pnum":."ff",...
20 22 70 63 6d 64 22 3a 20 22 33 66 22 2c 0d 0a ."pcmd":."3f",..
20 20 22 68 77 70 69 64 22 3a 20 22 66 66 66 66 .."hwpid":."ffff
22 0d 0a 7d                                     "..}

27-10-2017 03:07:09.736748 {DBG} "DpaHandler.cpp" ln:48 SendDpaMessage()
...

spinarr, 2017-10-27:
It also looks that json request does not contain response filed

Received from MQTT:
7b 0d 0a 20 20 22 63 74 79 70 65 22 3a 20 22 64 {...."ctype":."d
70 61 22 2c 0d 0a 20 20 22 74 79 70 65 22 3a 20 pa",...."type":.
22 72 61 77 2d 68 64 70 22 2c 0d 0a 20 20 22 6d "raw-hdp",...."m
73 67 69 64 22 3a 20 22 31 22 2c 0d 0a 20 20 22 sgid":."1",...."
6e 61 64 72 22 3a 20 22 30 31 22 2c 0d 0a 20 20 nadr":."01",....
22 70 6e 75 6d 22 3a 20 22 66 66 22 2c 0d 0a 20 "pnum":."ff",...
20 22 70 63 6d 64 22 3a 20 22 33 66 22 2c 0d 0a ."pcmd":."3f",..
20 20 22 68 77 70 69 64 22 3a 20 22 66 66 66 66 .."hwpid":."ffff
22 0d 0a 7d "..}

and there is response filed in json response message.
Response filed should be included only when requested in json request.

fmikulu, 2017-10-28:
Response filed should be included only when requested in json request is solved by dedicated issue: #82

fmikulu, 2017-10-28:
Please zip and add:
 - complete log file with described behaviour if possible.
 - used iqrf-daemon configuration
In case of other experiments use e.g. increasing **msgid** to simply distinguish incoming requests

spinarr, 2017-10-30:
I will try to recreate it with fresh log (or can be easily recreated when debugging).

At this moment in the log:

...
Going to sleep for: seconds="64"
30-10-2017 15:15:47.808272 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
30-10-2017 15:15:47.808658 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="64"
30-10-2017 15:16:51.809120 {DBG} "MqttMessaging.cpp" ln:293 connectThread()
Connecting: m_mqttBrokerAddr="tcp://iot.eclipse.org:1883" m_mqttClientId="IqrfDpaMessaging2"
30-10-2017 15:16:51.809445 {DBG} "MqttMessaging.cpp" ln:301 connectThread()
Going to sleep for: seconds="64"
... (repeats only this, nothing else is seen)

Sending from Paho web client:

{
    "ctype":"dpa",
    "type":"raw-hdp",
    "msgid":"1",
    "nadr":"01",
    "pnum":"ff",
    "pcmd":"3f",
    "hwpid":"ffff"
  }

Receiving:

Iqrf/DpaResponse  {     "ctype": "dpa",     "type": "raw-hdp",     "msgid": "1",     "nadr": "01",     "pnum": "ff",     "pcmd": "3f",     "hwpid": "ffff",     "rcode": "00",     "dpaval": "00",     "response": "",     "status": "ERROR_TIMEOUT" }

How is Iqrf/DpaRequest and Iqrf/DpaResponse handled when daemon paho client is not connected
or daemon is connected, but why are above messages in the daemon log and there are not 
messages regarding received DpaRequest and sending DpaResponse from and to Web client?

spinarr, 2017-10-30:
Configuration:

{
            "Name": "MqttMessaging2",
            "Enabled": true,
            "Properties": {
                "BrokerAddr": "tcp://iot.eclipse.org:1883",
                "ClientId": "IqrfDpaMessaging2",
                "Persistence": 1,
                "Qos": 1,
                "TopicRequest": "Iqrf/DpaRequest",
                "TopicResponse": "Iqrf/DpaResponse",
                "User": "",
                "Password": "",
                "EnabledSSL": false,
                "KeepAliveInterval": 20,
                "ConnectTimeout": 5,
                "MinReconnect": 1,
                "MaxReconnect": 64,
                "TrustStore": "server-ca.crt",
                "KeyStore": "client.pem",
                "PrivateKey": "client-privatekey.pem",
                "PrivateKeyPassword": "",
                "EnabledCipherSuites": "",
                "EnableServerCertAuth": true
            }
        }

fmikulu, 2017-11-05:
@spinarr I need complete configuration, please zip whole configuration directory and attache the file here. I would like to try the same on my machine

fmikulu, 2017-11-05:
Regarding ghost messages as you reported: _"How is Iqrf/DpaRequest and Iqrf/DpaResponse handled when daemon paho client is not connected or daemon is connected"_

Are you sure the only one instance of daemon (you watch) is running? What if other daemon responds? Is it possible to verify IP address or something else? For such problems we will have unique daemon id as described by: #71 

Meanwhile you can use different pair of topics, for tested instance wit paho web, e.g:
```
        "TopicRequest": "Iqrf/DpaRequest-TEST",
        "TopicResponse": "Iqrf/DpaResponse-TEST",
```

Note every MQTT client subscribed to MQTT server gets the topic.

spinarr, 2017-11-13:
Different topics have been tested and everything works nicely with them.

closed 2017-12-15 by spinarr