iqrfsdk/iqrf-daemon #87: Json Dpa Raw-Hdp

spinarr, 2017-12-19:
- If there is dpa response rcode is filled according to response
- If there is no dpa response rcode stays empty

fmikulu, 2017-12-19:
What does it mean empty? According http://www.json.org/ it is possible to set:
```
"rcode": null 
```
I thing this is not valid
`"rcode":`
What about some other number?
e.g.
`"rcode": -1`
or
Just to add numeric representation of result:
`"nstatus": number`
other then success would mean rcode and friends are invalid regardless their values.
BTW status is taken from dpa lib. Textual representation is there to have human readable result. If it is processed by SW in upper layers, then it doesn't make sense of course. It could be optional as some other values and `nstatus` mandatory. See conversion function from clibdpa:
```cpp
std::string DpaTransactionTask::getErrorStr() const
{
  switch (m_error) {
  case -4:
    return "ERROR_IFACE";
  case -3:
    return "ERROR_ABORTED";
  case -2:
    return "ERROR_PROMISE_TIMEOUT";
  case -1:
    return "ERROR_TIMEOUT";
  case STATUS_NO_ERROR:
    return "STATUS_NO_ERROR";
  case ERROR_FAIL:
    return "ERROR_FAIL";
  case ERROR_PCMD:
    return "ERROR_PCMD";
  case ERROR_PNUM:
    return "ERROR_PNUM";
  case ERROR_ADDR:
    return "ERROR_ADDR";
  case ERROR_DATA_LEN:
    return "ERROR_DATA_LEN";
  case ERROR_DATA:
    return "ERROR_DATA";
  case ERROR_HWPID:
    return "ERROR_HWPID";
  case ERROR_NADR:
    return "ERROR_NADR";
  case ERROR_IFACE_CUSTOM_HANDLER:
    return "ERROR_IFACE_CUSTOM_HANDLER";
  case ERROR_MISSING_CUSTOM_DPA_HANDLER:
    return "ERROR_MISSING_CUSTOM_DPA_HANDLER";
  case ERROR_USER_TO:
    return "ERROR_USER_TO";
  case STATUS_CONFIRMATION:
    return "STATUS_CONFIRMATION";
  case ERROR_USER_FROM:
  default:
    std::ostringstream os;
    os << std::hex << m_error;
    return os.str();
  }
}
```

spinarr, 2017-12-19:
If there is no dpa response:

"nadr":""
"pnum":""
"pcmd":""
"hwpid":""
"dpaval":""
"rcode":""

It is same principle as we have it in dpa raw structure.

commit 0a773693 (2017-12-21): Unresponded JSON raw-hdp reguest gets empty nadr, pnum, pcmd, hwpid, dpaval, rcode in JSON response fix #87
closed 2017-12-21 by fmikulu