iqrfsdk/iqrf-daemon #92: enable/disable KON-RASP-01 ENABLE pin GPIO23 on Raspberry Pi with Python

mikostn, 2018-02-07:
Hi,
this issue is related to #80 and iqrfsdk/clibspi#5 but I'm not sure what exactly "configurable mapping of pins" stands for.

I'm working on "reset coordinator" function in python.
It could be very easy done with this code for example:

```python
>>> from gpiozero import *
>>> en = OutputDevice(23, initial_value=True)
>>> en.off()
>>> en.on()
>>> en.close()
```
but as GPIO23 is used and set by: [clibspi/include/machines_def](https://github.com/iqrfsdk/clibspi/blob/master/include/machines_def.h) I'm afraid of conflicts on GPIO header.
Is it save to use other GPIO libs and take over control of pins (except for SPI0) when iqrf-daemon is running?
Or maybe clibspi or daemon provides some built-in function I could use?

I could probably use [iqrfsdk/clibspi/examples/io/io_example_reset/io_example_reset.c](https://github.com/iqrfsdk/clibspi/blob/master/examples/io/io_example_reset/io_example_reset.c) but if it's possible I want to avoid compiling special program just for that.

spinarr, 2018-02-07:
Quick question: can you reset coordinator via DPA command? Via MQ/MQTT?
https://www.iqrf.org/DpaTechGuide/#3.4.4%20Restart

mikostn, 2018-02-07:
Indeed I could but I'm looking fore something like "Hard Reset" solution.

Plus, this question also applies to LED (GPIO22) and BUTTON (GPIO7) also initiated by clibspi.
I'd like to take control on those pins and I just wonder if they are only inited and then "released" or "occupied" by clibspi or daemon and used later.

If occupied taking control of those pins might cause problems...

spinarr, 2018-02-07:
clibspi has a reference to 23 pin. 
spi_iqrf_init could be modified to actually restart coordinator during init.
Restarting daemon would restart coordinator as well.

mikostn, 2018-02-08:
Yes, it would do the job.
In general I think GPIO number and if clibspi or daemon controls or not it could be "configurable".
This could simplify different hardware designs.

I did not found any references (except heder file) to other pins. I assume they are not initiated and not used by clibspi or daemon. Is this right?

spinarr, 2018-02-08:
Yes, there are not used, only SPI lines and ENABLE.

mikostn, 2018-02-08:
Ok, grate! I've discussed restarting coordinator along with daemon, and this would be fine.

spinarr, 2018-03-01:
Restart TR when daemon boots, done in master.
Clibspi modified accordingly.

closed 2018-03-01 by spinarr