iqrfsdk/iqrf-daemon #96: IQRF Daemon SIGSEV crash on arm64

pepa-cz, 2018-04-23:
IQRF Daemon successfully started but any attempt with communication fails. When DPA request is sent through MQTT or MQ Channel, daemon crashes on segmentation fault.
```
root@debian:~/iqrf/iqrf-daemon/scripts/py# /usr/bin/iqrf_startup /etc/iqrf-daemon/config.json

/usr/bin/iqrf_startup
============================================================================
DAEMON_VERSION="v1.1.0" BUILD_TIMESTAMP="2018-04-23 13:22:29"
============================================================================
Loading configuration file: cfgFileName="/etc/iqrf-daemon/config.json"
gpio_setup() setDir success: 1
IQRF Gateway Daemon started .....
Segmentation fault
```

- IQRF Daemon built from source `make build-debug && make install`
- Running architecture (not in official IQRF repo) `Linux debian 4.11.2 #4 SMP Wed Jul 19 17:53:37 CST 2017 aarch64 GNU/Linux`

Valgrind info:

```
root@debian:~/iqrf/iqrf-daemon/scripts/py# /usr/bin/iqrf_startup /etc/iqrf-daemon/config.json

/usr/bin/iqrf_startup
============================================================================
DAEMON_VERSION="v1.1.0" BUILD_TIMESTAMP="2018-04-23 13:22:29"
============================================================================
Loading configuration file: cfgFileName="/etc/iqrf-daemon/config.json"
gpio_setup() setDir success: 1
IQRF Gateway Daemon started .....
Segmentation fault
root@debian:~/iqrf/iqrf-daemon/scripts/py# make build-debug^C
root@debian:~/iqrf/iqrf-daemon/scripts/py# valgrind --show-reachable=yes iqrf_startup /etc/iqrf-daemon/config.json
==3265== Memcheck, a memory error detector
==3265== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==3265== Using Valgrind-3.12.0.SVN and LibVEX; rerun with -h for copyright info
==3265== Command: iqrf_startup /etc/iqrf-daemon/config.json
==3265==

iqrf_startup
============================================================================
DAEMON_VERSION="v1.1.0" BUILD_TIMESTAMP="2018-04-23 13:22:29"
============================================================================
Loading configuration file: cfgFileName="/etc/iqrf-daemon/config.json"
gpio_setup() setDir success: 1
IQRF Gateway Daemon started .....
==3265== Thread 4:
==3265== Invalid read of size 4
==3265==    at 0x266924: DpaTransfer::CheckTimeout() (DpaTransfer.cpp:359)
==3265==    by 0x266633: DpaTransfer::ProcessStatus() (DpaTransfer.cpp:349)
==3265==    by 0x26743B: DpaTransfer::IsInProgress() (DpaTransfer.cpp:407)
==3265==    by 0x25BBD3: DpaHandler::IsDpaMessageInProgress() const (DpaHandler.cpp:92)
==3265==    by 0x25AF17: DpaHandler::SendDpaMessage(DpaMessage const&, DpaTransaction*) (DpaHandler.cpp:50)
==3265==    by 0x25D217: DpaHandler::ExecuteDpaTransaction(DpaTransaction&) (DpaHandler.cpp:247)
==3265==    by 0x2108F7: DaemonController::executeDpaTransactionFunc(DpaTransaction*) (DaemonController.cpp:67)
==3265==    by 0x215293: DaemonController::startIqrfIf()::{lambda(DpaTransaction*)#1}::operator()(DpaTransaction*) co                                             nst (DaemonController.cpp:417)
==3265==    by 0x21F587: std::_Function_handler<void (DpaTransaction*), DaemonController::startIqrfIf()::{lambda(DpaT                                             ransaction*)#1}>::_M_invoke(std::_Any_data const&, DpaTransaction*&&) (functional:1731)
==3265==    by 0x226813: std::function<void (DpaTransaction*)>::operator()(DpaTransaction*) const (functional:2127)
==3265==    by 0x224423: TaskQueue<DpaTransaction*>::worker() (TaskQueue.h:130)
==3265==    by 0x22E577: void std::__invoke_impl<void, void (TaskQueue<DpaTransaction*>::* const&)(), TaskQueue<DpaTr                                             ansaction*>*>(std::__invoke_memfun_deref, void (TaskQueue<DpaTransaction*>::* const&)(), TaskQueue<DpaTransaction*>*&                                             &) (functional:227)
==3265==  Address 0x58 is not stack'd, malloc'd or (recently) free'd
==3265==
==3265==
==3265== Process terminating with default action of signal 11 (SIGSEGV)
==3265==  Access not within mapped region at address 0x58
==3265==    at 0x266924: DpaTransfer::CheckTimeout() (DpaTransfer.cpp:359)
==3265==    by 0x266633: DpaTransfer::ProcessStatus() (DpaTransfer.cpp:349)
==3265==    by 0x26743B: DpaTransfer::IsInProgress() (DpaTransfer.cpp:407)
==3265==    by 0x25BBD3: DpaHandler::IsDpaMessageInProgress() const (DpaHandler.cpp:92)
==3265==    by 0x25AF17: DpaHandler::SendDpaMessage(DpaMessage const&, DpaTransaction*) (DpaHandler.cpp:50)
==3265==    by 0x25D217: DpaHandler::ExecuteDpaTransaction(DpaTransaction&) (DpaHandler.cpp:247)
==3265==    by 0x2108F7: DaemonController::executeDpaTransactionFunc(DpaTransaction*) (DaemonController.cpp:67)
==3265==    by 0x215293: DaemonController::startIqrfIf()::{lambda(DpaTransaction*)#1}::operator()(DpaTransaction*) co                                             nst (DaemonController.cpp:417)
==3265==    by 0x21F587: std::_Function_handler<void (DpaTransaction*), DaemonController::startIqrfIf()::{lambda(DpaT                                             ransaction*)#1}>::_M_invoke(std::_Any_data const&, DpaTransaction*&&) (functional:1731)
==3265==    by 0x226813: std::function<void (DpaTransaction*)>::operator()(DpaTransaction*) const (functional:2127)
==3265==    by 0x224423: TaskQueue<DpaTransaction*>::worker() (TaskQueue.h:130)
==3265==    by 0x22E577: void std::__invoke_impl<void, void (TaskQueue<DpaTransaction*>::* const&)(), TaskQueue<DpaTr                                             ansaction*>*>(std::__invoke_memfun_deref, void (TaskQueue<DpaTransaction*>::* const&)(), TaskQueue<DpaTransaction*>*&                                             &) (functional:227)
==3265==  If you believe this happened as a result of a stack
==3265==  overflow in your program's main thread (unlikely but
==3265==  possible), you can try to increase the size of the
==3265==  main thread stack using the --main-stacksize= flag.
==3265==  The main thread stack size used in this run was 8388608.
==3265==
==3265== HEAP SUMMARY:
==3265==     in use at exit: 974,518 bytes in 3,383 blocks
==3265==   total heap usage: 15,215 allocs, 11,832 frees, 1,952,973 bytes allocated
==3265==
==3265== LEAK SUMMARY:
==3265==    definitely lost: 0 bytes in 0 blocks
==3265==    indirectly lost: 0 bytes in 0 blocks
==3265==      possibly lost: 3,168 bytes in 11 blocks
==3265==    still reachable: 971,350 bytes in 3,372 blocks
==3265==         suppressed: 0 bytes in 0 blocks
==3265== Rerun with --leak-check=full to see details of leaked memory
==3265==
==3265== For counts of detected and suppressed errors, rerun with: -v
==3265== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Killed
```

pepa-cz, 2018-04-23:
The same situation also occurs also in case the daemon is stopped in a usual way by pressing CTRL+C
```
root@debian:~/iqrf/iqrf-daemon/scripts/py# /usr/bin/iqrf_startup /etc/iqrf-daemon/config.json

/usr/bin/iqrf_startup
============================================================================
DAEMON_VERSION="v1.1.0" BUILD_TIMESTAMP="2018-04-23 13:22:29"
============================================================================
Loading configuration file: cfgFileName="/etc/iqrf-daemon/config.json"
gpio_setup() setDir success: 1
IQRF Gateway Daemon started .....
^C
..... Normal exit.
Segmentation fault
```

fmikulu, 2018-04-28:
There is a bug indeed. The SW shall be ready for it. But bad behavior starts with:
 
```
Opened trace file: m_traceFileName="/var/log/iqrf-daemon.log" m_traceFileSize="10485760" 
23-04-2018 14:45:33.756471 {INF} "DaemonController.cpp" ln:368 startIqrfIf()
m_iqrfInterfaceName="/dev/spidev0.0" 
23-04-2018 14:45:34.562386 {DBG} "IqrfSpiChannel.cpp" ln:194 listen()
{ENTER} thread starts
23-04-2018 14:45:34.563144 {DBG} "IqrfSpiChannel.cpp" ln:197 listen()
SPI is ready
23-04-2018 14:45:34.574521 {INF} "IqrfSpiChannel.cpp" ln:127 getState()
SPI status1: spiStatus1.dataNotReadyStatus="0" 
23-04-2018 14:45:34.575008 {INF} "IqrfSpiChannel.cpp" ln:128 getState()
SPI status2: spiStatus2.dataNotReadyStatus="0" 
23-04-2018 14:45:34.688361 {INF} "IqrfSpiChannel.cpp" ln:127 getState()
SPI status1: spiStatus1.dataNotReadyStatus="0" 
23-04-2018 14:45:34.688852 {INF} "IqrfSpiChannel.cpp" ln:128 getState()
SPI status2: spiStatus2.dataNotReadyStatus="0" 
...
23-04-2018 14:45:35.714243 {ERR} "DaemonController.cpp" ln:408 startIqrfIf()
IQRF interface is not ready ...
23-04-2018 14:45:35.715427 {INF} "DpaHandler.cpp" ln:31 DpaHandler()
Ctor default user timeout: m_defaultTimeoutMs="200" 
23-04-2018 14:45:35.715902 {DBG} "DpaTransfer.cpp" ln:345 ProcessStatus()
{ENTER} 
23-04-2018 14:45:35.716481 {INF} "DpaTransfer.cpp" ln:360 CheckTimeout()
Transfer status: created
23-04-2018 14:45:35.716860 {DBG} "DpaTransfer.cpp" ln:351 ProcessStatus()
{LEAVE} m_status="0" 
23-04-2018 14:45:35.718296 {DBG} "DpaHandler.cpp" ln:223 ExecuteDpaTransaction()
Using default timeout: defaultTimeout="200" 
23-04-2018 14:45:35.718783 {DBG} "DpaHandler.cpp" ln:48 SendDpaMessage()
{ENTER} 
23-04-2018 14:45:35.719212 {DBG} "DpaTransfer.cpp" ln:345 ProcessStatus()
{ENTER} 
23-04-2018 14:45:35.719577 {INF} "DpaTransfer.cpp" ln:360 CheckTimeout()
Transfer status: created
23-04-2018 14:45:35.719943 {DBG} "DpaTransfer.cpp" ln:351 ProcessStatus()
{LEAVE} m_status="0" 
23-04-2018 14:45:35.720648 {DBG} "DpaHandler.cpp" ln:56 SendDpaMessage()
<<<<<<<<<<<<<<<<<<
Sent to DPA interface: 
00 00 02 00 ff ff                               ......          

23-04-2018 14:45:35.721238 {INF} "IqrfSpiChannel.cpp" ln:148 sendTo()
Sending to IQRF SPI: 
00 00 02 00 ff ff                               ......          

23-04-2018 14:45:35.721801 {DBG} "IqrfSpiChannel.cpp" ln:151 sendTo()
Trying to sent: 1.1
23-04-2018 14:45:35.723590 {DBG} "IqrfSpiChannel.cpp" ln:151 sendTo()
Trying to sent: 1.2
23-04-2018 14:45:35.725344 {DBG} "IqrfSpiChannel.cpp" ln:151 sendTo()
...
23-04-2018 14:45:35.736789 {WAR} "IqrfSpiChannel.cpp" ln:187 sendTo()
Cannot send to SPI: message is dropped
23-04-2018 14:45:35.739737 {WAR} "DpaHandler.cpp" ln:270 ExecuteDpaTransaction()
Send error occured: basic_string::_M_construct null not valid
23-04-2018 14:45:35.740736 {WAR} "DaemonController.cpp" ln:449 startDpa()
Throwing std::logic_error: Cannot get TR parameters
23-04-2018 14:45:35.742035 {ERR} "DaemonController.cpp" ln:462 startDpa()
There was an error during DPA handler creation: Cannot get TR parameters
```
Then there is the crash in next try to send Dpa message. I continue with analyze why it crash and I'll make it more robust to cope with it.