iqrfsdk/iqrf-daemon #97: IQRF daemon crash on arm7l

stradiot, 2018-10-04:
- IQRF Daemon successfully started and is able to connect to IQRF IDE, but crashes occasionally after throwing CDCImplException. 
- System architecture is `Linux LogicMachine 4.4.152 #0 PREEMPT Tue Aug 28 07:47:38 UTC 2018 armv7l GNU/Linux` 
- Daemon is in service mode, communicating with IQRF IDE using UDP channel. 

[image]
[image]
[log.txt](https://github.com/iqrfsdk/iqrf-daemon/files/2446946/log.txt)
[config.txt](https://github.com/iqrfsdk/iqrf-daemon/files/2446948/config.txt)

fmikulu, 2018-10-05:
The primary reason is an error on USB CDC interface towards IQRF coordinator. On the other the daemon shall provide a WAR trace and continue. Described behaviour indicates an untreated exception in CDC sending thread.

Do you use Release packages or are you able to build all from scratch?
I can provide a quick fix on master branch, just to keep daemon running in similar cases, however creating a new release packages may take a time.

stradiot, 2018-10-05:
We do not use release packages, we built it from scratch.
Thank you

fmikulu, 2018-10-05:
The fix is in referenced library : https://github.com/iqrfsdk/clibdpa/commit/58ba03e7ca0da9653114b544aa671dc1e5b466ad

It doesn't solve the runtime error on CDC. It just keeps the daemon running. 

Soon or later we disrupt support of this V1 library. Please consider test and usage of prepared V2, existing in beta now: https://github.com/iqrfsdk/iqrf-gateway-daemon. The release is planned at the end of Oct/2018

closed 2018-10-05 by stradiot